arkypita/LaserGRBL #1108: Changing LaserGRBL Zero Point

3DStyle, 2020-08-26:
Is it possible to change the laser zero point?
As default LaserGRBL will start at the bottom left, like this:
[image]

I want to have the control to change the zero point location, lets say I have a piece of wood that I want to laser a logo on its
center, so I'll be able to change the zero point in LaserGRBL to the center of the image, something that will look like this:

[image]

3DStyle, 2020-08-26:
Thanks for the tip Alex, it works great!
However, an implemented button in LaserGRBL that will do this job for you will be more convenient.

arkypita, 2020-08-29:
already in #64 as "Add image centring helper #90"
however i think you can create a custom button that set zero point at center of your job after import

closed 2020-12-16 by arkypita