arkypita/LaserGRBL #1110: Compilation error on Ubuntu due to file name case sensitivity

mgaman, 2020-08-27:
When building via 'xbuild LaserGRBL.sln' I got the error 

Got error CSC: error CS2001: Source file '/home/henryd/Documents/CS/LaserGRBL-master/LaserGRBL/UserControls/MyPictureBox.designer.cs' could not be found. --- NOTE lowercase designer

My immediate fix was to make a link to the existing file to a new file with the correct case for Designer

ln MyPictureBox.Designer.cs MyPictureBox.designer.cs

RafaelEstevamReis, 2020-08-29:
I have a DotNet core port-in-progress of LaserGRBL, my ultimate goal is to be Linux compatible

DotNet Core dot not use file-names on solution/project
Can you tri this on my branch ? https://github.com/RafaelEstevamReis/LaserGRBL

You can have a read on my PR thread to understand what am I changing

mgaman, 2020-09-02:
I just cloned your branch (to linux) and ran 'dotnet build' and got an error message that windows desktop apps can only be built on windows. As I understand it dotnet core is only for console or web apps.
How do you build it?

RafaelEstevamReis, 2020-09-02:
"As I understand it dotnet core is only for console or web apps."
Seems true on linux
I'm still on library and console part yet

If you could update the project files to be booth .Net Framework and .Net Core, I can help migrating back some libraries and it became more robust

But, this point foward, we can no longer discuss it here
Can you repost your issue on My branch ?

mgaman, 2020-09-03:
There is no 'issues' tab on your fork of the project.
You need to go into Settings and enable Issues

mgaman, 2020-09-03:
Sorry, wrong repository

RafaelEstevamReis, 2020-09-03:
Ops, my bad, I didn't even notice that
Now it has