arkypita/LaserGRBL #1111: adding stops for large programs

GAMKCMO, 2020-08-29:
Some times I have projects to large for my 3018 pro.  Requires me to run 2 separate programs and reposition the work piece. Moving the work is not a problem,  Splitting the program and reloading them is a pain and can create issues.  Is it possible to run one file and put a "stop/Continue" command in so the 3018 would stop/pause at a specific spot. Then move the work piece and continue?

BeatSlayer, 2020-08-29:
It's just way easier if you just cut your image, and do the engraving with the resulting images, you can even add some alignment features depending on the geometry complexity of the repositionings, like angling.

GAMKCMO, 2020-08-29:
Thanks for the response,  Not sure I follow what your telling me.   Making  cribbage boards.  Use easel for the holes and it easy to align second group of holes.  Using GRBL to burn the numbers and text on.  GRBL is not that easy to align in the program when you do the second file.   Would like to use 1 file and stop it half way tru, move the board with low laser on for alignment, then hit continue.   I guess I could do manually but would have to watch it for couple hours to time my stop. I'm planning on doing a Number of these (gifts) and want to simplify the process, and decrease the chance of error [since I'm doing this on a board that I have already invested a lot time in preparing the holes.

BeatSlayer, 2020-08-29:
If you're working with multiple objects I would advise making a jig.
I would put some board beneath the work with some alignment features, and use those to align the work to the laser.

arkypita, 2020-08-29:
not sure this is what you ask for, but you can manually edit a gcode file with all your parts and insert M226 command between each part.

M226 is a pause command that stop grbl processing next commands until you press resume button

Ps: i am not sure that M226 is the right code.

pelowitz, 2020-08-29:
Although I build my own laser engravers or cnc machines at any size I want (latest cnc has a working field of about 60"x60") it isn't expensive to build a laser system that can do your entire cribbage board.  My laser systems all run on timing belts instead of ball screws, but my CNC systems all use ball screws.  I solve your problem by creating two or more gcode files, each covering a different portion of the working area.  I always add alignment points so that when I move the target material I know the alignment is correct.  This procedure is extremely easy when using tools like vectric's Carve2D.

GAMKCMO, 2020-08-30:
Arkypita,  That sound like what I'm after.  Could't find the M226 but the below. Does M00 sound right?
[image]
Now I'm trying to find and editor to insert the code.  Working with a couple free options,  Note++,  and NC Viewer. Any suggestions? One problem seems to lead to another.

pelowitz, 2020-08-30:
GAMKCMO-the gcode files are simple text files.  In windows Notepad works fine.  The only hard part is discovering where to place your stop.

GAMKCMO, 2020-09-01:
Making some progress.  Using NCviewer was able to run and view code.  Found the spot I wanted to insert a stop/pause.  Inserted M00 in that place and reran the program again in viewer and it went on by and didn't stop.  Also tried M01 and M226 with no difference.  Any thoughts or suggestions?  
Thanks for your help. 
 Below is the snip of the place I used the M226 code, I'm I using it right?
[image]

pelowitz, 2020-09-01:
M0 as well as M00 works well in GRBL version 1.1f to halt the program.  Do a $i to check which version you are running, assuming you are using the GRBL firmware.

gmmanonymus111, 2020-09-01:
I'm not sure about this, but according to the GCODE Wiki [here](https://wiki.shapeoko.com/index.php/G-Code), you should use `M0` (or `M2` on older machines). `M00` and `M226` might be invalid

GAMKCMO, 2020-09-01:
I have GRBL v3.7.1 just updated it. Not sure about firmware and where to check.  The machine is a 3018pro about one year old.  I'm not testing the M commands on the 3018pro.  I'm testing it with the "NC viewer" and running it from a point before the M code and it it doesn't stop.  I assuming the NC viewer would show the same results as it plots to file.
Seems like this should be simple and I'm likely doing something wrong?
Thanks for your help
Tried to attach file but it wouldn't take a   .nc

pelowitz, 2020-09-01:
LaserGRBL is a great program that can create g-code and send it (or another g-code file from another program) to a machine such as the 3018pro or basically any machine that interprets g-code to control the positioning.  The most commonly used program (firmware) running **in** hobby machines is called grbl (hence the name LaserGRBL).  You can type commands directly to your machine by opening something like hypertrm on widows, setting the baud rate to the same as LaserGRBL is using and open the same port as it uses.  Once connected type $I and hit enter (or send depending on which program you are using.  LaserGRBL also does this for you when it first connects to your machine.  To do this manually in LaserGRBL add a macro button that sends the $I command (that is an i not an L). You can delete the button later.  I recommend you run grbl 1.1f as some people have had a problem with 1.1h.  The author of LaserGRBL considered the need to "flash" a newer version of the firmware in your machine and has provided a menu selection to accomplish this task.  Prior to doing it you'll need to download version 1.1f (if you are not already running it).  Do a search on the web and get it from the repository.

StuartB4, 2020-09-01:
` Prior to doing it you'll need to download version 1.1f (if you are not already running it). Do a search on the web and get it from the repository.`

1.1f is included in LaserGRBL on the Tools menu, no need to download it seperately.

[image]

pelowitz, 2020-09-01:
Hay, fantastic.  I had never clicked on the dropdowns 'cause I never needed to.  Thanks.

GAMKCMO, 2020-09-02:
I will give that a try,   I have seen the pull down before but the WARNING  scared me.  after this should I open the file in question and save it through Laser GRBL,  then try in viewer again?  Thanks.

pelowitz, 2020-09-02:
Before you load the firmware check which version of firmware you are currently running (send $I to it).  Reloading firmware may not be the problem here.  The question is "What version is you machine running now?"  Then you want to do a bit of web searching to verify that version didn't support M0.  I recommend you do this prior to updating the firmware.  Yes, there is risk in that there are things that can go wrong and brick your hardware.  Although, in loading dozens of times on other hardware as well as my 3018ProVer I've never had a problem that was unsolvable.

GAMKCMO, 2020-09-02:
OK I think I figured it out.   None of the stop options work in the NCviewer.   I edited the file in a notes and ran it on the 3018 .   Place the edit close to the beginning just to test.
M3 S30    put lase in low power for alignment purposes
M0           Stop/hold

after repositioning work piece with laser in low,  simple hit the running man and off it goes!    
Now I still need to find the line number to edit at.   Any way to display the line # in notes?  59818 is the number in NC viewer.   

Thanks for everyone help!   
Think this is closed

BeatSlayer, 2020-09-02:
Notepad++ has line numbers and go-to line feature

closed 2020-09-05 by GAMKCMO