arkypita/LaserGRBL #1112: Option to turn on laser for focusing

MrProgrammer69, 2020-08-29:
I had to focus my laser while it was engraving.

Can we have an option to turn on the laser so we can focus it on our material please.

Sorry if this is already available, but have only just had the laser/software and could not see the option.

gmmanonymus111, 2020-08-29:
There should be a custom button pack [here](https://github.com/arkypita/LaserGRBL/issues/306#issuecomment-653736075)  which you might find useful. It contains buttons to focus/position your laser, preview your engraving area etc.

StuartB4, 2020-08-29:
This link will take you to the download for the buttons. Just scroll down to the bottom and click the Download button.
When you have the file start lasergrbl and click connect. Right click on an empty part of the bar at the bottom of the main window, next to the little man icon. Click Import custom buttons and then select the file you downloaded. No need to extract it, then you will get an option to install all the buttons seperately, just click Yes or No to the ones you want or dont want.
http://lasergrbl.com/usage/custom-buttons

MrProgrammer69, 2020-08-30:
Got the buttons thanks, really helpful when laying out work and testing.
When focusing the laser what should I bee looking for, I thought it was focused, but when it started, it did not engrave, I had to focus live until it started to burn the MDF :-<

StuartB4, 2020-08-30:
You just need the tiniest dot possible. All laser dots are actually rectangle so you won't get a perfect dot, but you can get pretty close to it.
I use a USB microscope to focus, cost me £12 UK.  It plugs in to a USB port and you aim it at the laser on low power and focus the dot on the screen first with the microscope, then focus the dot with the laser lens while watching the PC screen. You don't need the protective glasses on to do it either.  You can see it here in my video. https://www.youtube.com/watch?v=ebKNe25McdU&feature=youtu.be

MrProgrammer69, 2020-09-01:
StuartB4, I had what I thought was the tiniest dot possible, and was really pleased till it didn't burn :-)

I have one of those microphones, so I will give that a go and see how I go.
How can I turn the power down to stop it scorching it as much as it does.

StuartB4, 2020-09-01:
Do you mean when it's engraving? If so you can reduce the power in Target image settings by entering a lower number in S-MAX box,  Also use M4 for engraving M3 is better for cutting.

If you get the focus sorted out then you should get a bit less scorching providing the power is not to high, it also depends on what material you are engraving / cutting.  Also have a look at Air assist, there are videos on youtube it helps a lot especailly for cutting, and it gives you less scorching.

MrProgrammer69, 2020-09-02:
I've been trying MDF as I had a massive sheet that I've cut up. Trouble is the burning small in the house, Mrs wasn't happy :-)
I need to make an enclosure with a extractor/filter and soundproofing for engraving!

I have brought some 3mm Plywood and also some foam board, but it just melted that :-(
Now you get onto cutting, I think my laser is only 450mw so will it cut? or is it a case of doing multiple passes until it has?

StuartB4, 2020-09-03:
450 mW will probably cut paper and thin card but not much more. You will nead something like a 5 watt or more to do any serious cutting, and then really slow speed and multiple passes. I have a 6 watt output and cut 3mm laser ply with speed of 150 mm's/min at 80 to 90 percent power and 3 to 5 passes depending on the quality of the plywood. Also have a look on youtube at Air assist. It's a big help when cutting.

closed 2020-12-16 by arkypita