arkypita/LaserGRBL #1113: Output scale is wrong

batusekerci, 2020-08-31:
Hello everyone,

My problem is when I try to print something, output scale is wrong. I am trying to engrave an image, in the program measurements are  210 x 130 mm in the uploading screen , 192 x 120 mm in the main screen but the output is like 100 x 70 mm. 

Am I doing something wrong? Should I use Autosize option or it is something else with grbl? 
By the way, problem is occured when I have updated grbl 0.9j to 1.1h and also tried to use another program.
If the problem is about update, can I revert the changes by downgrading?

StuartB4, 2020-08-31:
It could be your steps/mm in grbl configuration. The settings are on the Grbl menu, grbl configuration.
Make sure lasergrbl is connected first.
You can check them by looking at the list for $100 and $101. write down the numbers in the value boxes. 
Then have a look at this website. It will let you enter your current numbers, jog the machine, and then measure how much the machine moved and enter that number on the website. Then it will calculate the correct numbers that you can put in to grbl $100 and $101, after entering the numbers in grbl you click the Write button to store them.
https://jscalc.io/calc/tHjfIiKMKjDF42Hg?fbclid=IwAR219p6S0qWzVAsY8aLC8nyQaXaMpfa-mSTNCIXA7TAMPa3VzrGoCDcimsI

Like this.
[image]

batusekerci, 2020-08-31:
Thanks, it worked like a charm!

Last question: Can we test the laser someway? I mean for a second for example, could laser be on?

BeatSlayer, 2020-08-31:
Give a try to this [custom buttons](https://lasergrbl.com/usage/custom-buttons/)

StuartB4, 2020-08-31:
This link will take you to the download for the buttons. Just scroll down to the bottom and click the Download button.
When you have the file start lasergrbl and click connect. Right click on an empty part of the bar at the bottom of the main window, next to the little man icon. Click Import custom buttons and then select the file you downloaded. No need to extract it, then you will get an option to install all the buttons seperately, just click Yes or No to the ones you want or dont want.
http://lasergrbl.com/usage/custom-buttons

batusekerci, 2020-09-01:
Thanks everyone!

closed 2020-09-01 by batusekerci