arkypita/LaserGRBL #1116: Ortur laser master 2 firmware

kubala1669, 2020-09-04:
please help, i need firmware for laser master 2

StuartB4, 2020-09-05:
Join the Ortur users Facebook group. 
Ask Gil Araújo, he usually has all the answers for the Otur machine.
https://www.facebook.com/groups/orturusergroup

BeatSlayer, 2020-09-05:
Can you share the info that you find about it?

I don't feel like making a facebook profile to update a firmware.

StuartB4, 2020-09-05:
Just found this on the official Ortur FB site. 

Ortur Laser Master 1 Firmware links.
http://s000.tinyupload.com/?file_id=20540506184393439063

Ortur Laser Master 2
http://s000.tinyupload.com/?file_id=89707801244192121779

HOW to Update your Firmware:
Please connect the Ortur Laser Master Laser Engraver to the Windows OS computer by USB cable and follow the steps below.
Note: Be sure Lightburn, laserGrbl or any other software that communicates to the USB port are not running.
a) Press and hold power button
b) Press the Reset Button one time
c) In your computer a folder will show up (this folder is just as similar as a flash driver inserted into the computer)
d) Release the power button.
e) Copy the OLF_134.BIN in the folder
f) Wait a few seconds, then press Reset Button again

The #firmware should now be upgraded to the latest version.

arkypita, 2020-09-05:
In package you can find a link to download manual, software, guides and contains firmware and firmware install info

Lajos343, 2020-12-10:
HI I would need an Ortur laser master2 Bin file
STM32F103C8T6 Ic 'empty' and I would upload it with Wellon VP998 Programmer
Please help!

Lajos343, 2020-12-10:
I did not connect the cables properly so the STM32F103C8T6 failed to replace it and now the STM32F103C8T6 is empty missing the 'BIN file'.
Can't find a solution to this?
Lajos

closed 2020-12-16 by arkypita