arkypita/LaserGRBL #1117: Upgrade for the start/end of engraving line

Max59570, 2020-09-08:
Hello,

I use LaserGRBL with a small CNC machine with a laser. I have a problem at the end and at the beginning of the engraving line. The quality is not good, I think is due to the acceleration and deceleration of the machine. 

Is there a parameter for LaserGRBL to activate the laser only in the same speed. Like this the machine start before the image to engrave and when it arrives to the image the speed is OK and always the same (same for the end of line, the machine starts to stop the mouvement after the laser is OFF).

Thank you for you help.
BR
Max

distebia, 2020-09-08:
hai provato a mettere M4 al posto di M3?

StuartB4, 2020-09-08:
Do you have Laser mode enabled in grbl configuration. 
Start LaserGRBL and click Connect.
Then go to the Grbl menu and click Grbl configuration you will get
the list of settings. Look down the list and make sure $30 is set to 255.
and $32 is set to 1. If they are not change them and then click the Write button and Close.

Max59570, 2020-09-09:
Hello, I tried to put M4 instead of M3 and the result is better! Thank you 👍 

In which case we should use M3 and which case we should use M4?

arkypita, 2020-09-09:
https://github.com/arkypita/LaserGRBL/issues/1103#issuecomment-684027981

fude57, 2020-09-24:
habe das selbe problem auch mit m4 , kann mir jemand weiter helfen, brennt anfang u ende immer stärker danke im voraus

closed 2020-12-16 by arkypita