arkypita/LaserGRBL #1118: Immediate start of laser and Z axis motor

Branes1, 2020-09-08:
I have a samll Chinese CNC 3018 machine. It works good with other software but with LaserGrbl (which I prefer for certain jobs) I have an issue.
When I press the Connect button the laser immediately turns on with full power and the Z motor starts to work and lowers the laser. The only way to stop this is to completely disconect the power and USB connection.
I tried with different computers and OSs (currently with WIN7/64, latest version of LaserGrbl), but only sometimes this problem does not show. 
Can somebody help?
Thank you in advance

arkypita, 2020-09-09:
First: try playing with soft-reset and hard-reset combination

[image]

Second: You may have startup blocks:

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#n---view-startup-blocks

Third: the version of grbl (the firmware) installed on CNC 3018 machine can be a customized version which is intended for use as a CNC. Try flashing a brand-new firmware. (you will lost any eventual customization, so you can't go back if seller doesn't give you their customized firmware)

http://lasergrbl.com/usage/flash/

Fourth: your board could be have negative logic for laser on/off. Standard is 5V on, 0V off.
You can make some hardware electrical changes/converter to invert logic, or you can re-compile grbl firmware by yourself to produce an inverted logic command.

Branes1, 2020-09-11:
Thank you for your ng answer. But, as I described earlier, the problem is
very essentia. Let me try to explain it again.

1. I switch ON the laptop WIN7/64
2. Start the LaserGrbl
3. Switch on the CNC
4. Connect the CNC USB connection (or it is already connected to the
computer)
5. Push the Connect button in LaserGrbl
And now the problem starts immediately:
The Laser on CNC lowers from Z-axis with a strange loud sound and the Laser
is ON full power (if I would let go it would go completely down to the CNC
table and wouldn't stop there.
There is no soft reset, no hard reset, the only thing that helps is to
disconnect USB from the computer and disconnect the power from the line.
Even the end switches do not help. If I disconnect only the power (24V)
from CNC electronics the Z motor stops but starts immediately if I put the
power on again. So I have to disconnect the USB also.

But, sometimes, very rarely and without obvious reason, the LaserGrbl works
fine. And, again, I tried all the versions of LaserGrbl on different
computers with different OSs.

I have to say that I would really like to use this software, because of its
simplicity and clean GUI, but...

Sincerely yours

StuartB4, 2020-09-11:
Do you have Laser mode enabled. I think you need to change $32 to 1 in grbl configuration on the grbl menu.

closed 2020-12-16 by arkypita