arkypita/LaserGRBL #1121: How to burn a photo from Photograv 3.0 without converting ?

qulet, 2020-09-12:
267/5000
Hello!
The Photograv program generates photos consisting of black points. How to burn such a photo with an LED laser without converting in the GRBL control? Is there a program that will generate G-Gode for GRBL from an image generated in Photograv 3.0?

StuartB4, 2020-09-12:
You don't need to use Photograv. In LaserGRBL you just load an image in and adjust Brightness and Contrast if you need to. Select the engraving method 1Bit BW dithering and select which type of dither you want, like Jarvis, Atkinson etc from the drop down list. Then set whether you want it to engrave Horizontally, Vertically or Diagonally, then set the Quality. The higher the number here the better quality but slower engraving.

qulet, 2020-09-12:
StuartB4
 Dziękuję za odpowiedź ...To wszystko o ustawieniach LaserGrbl WIEM ! Ale jakość ditheringu w programie Photograv jest o klasę  lepsza :))   i tę jakość  chciałbym  przenieść do G-code  bez dodatkowej konwersji...

BeatSlayer, 2020-09-12:
Get Photograv to produce GCode and just stream it with LaserGRBL.

qulet, 2020-09-13:
Mam wersje  Photograv 3.0.  Program   nie tworzy G-code,  generuje tylko obrazki 1-Bit  i tu tkwi problem....

BeatSlayer, 2020-09-13:
If Photograv doesn't generate GCode than LaserGRBL will have to do some form of conversion.

With the current options available, I believe the closest you'll get should be BW line2line conversion, but as @StuartB4 said you don't really need Photograv because LaserGRBL will do the same, or a even better job if you spend some time tuning your image settings, at converting those images.

qulet, 2020-09-13:
Diękuję wszystkim  za  porady !
Pozdrawiam !