arkypita/LaserGRBL #1123: Laser Vigotec

robywoodwork, 2020-09-15:
good morning, my name is Roberto.  I bought the vigotec laser model VGL x7 with Wi-Fi (which I would not use at all). My question is: I would like to put the firmware of laser GRBL 1.1f but if it does not work, how can I go back?  There is a way to extract the machine firmware before I try to change it in order to recover the card.  Thanks in advance for the help.  a smile, Roberto

StuartB4, 2020-09-15:
If you start LaserGRBL and click Connect what does it show in the left window?
[image]

robywoodwork, 2020-09-15:
Hi Stuart, I'm out of the house right now, I'll come back tonight and I'll tell you.  thank you so much in the meantime

StuartB4, 2020-09-15:
Ok.

robywoodwork, 2020-09-15:
[image]

robywoodwork, 2020-09-15:
This si the motherboard

robywoodwork, 2020-09-15:
this is what GRBL tells me and in the tools menu the various options for the firmware at my own risk.  I love GRBL, I have been using it for a few months with the Eleksmaker laser, it would be great to be able to use it with my new 20w Vigo laser too.  hope you can help me.  a smile, Roberto

robywoodwork, 2020-09-15:
[image]

robywoodwork, 2020-09-15:
[image]

robywoodwork, 2020-09-15:
[image]

robywoodwork, 2020-09-15:
[image]

StuartB4, 2020-09-16:
It should work because the main chip on the board is an Atmel chip which is what the Arduinos use. Select Nano on the drop down menu. The baud rate should set it's self to the correct one. If you do flash the firmware you may need to install the driver from the Tools menu as well. The other 2 chips underneath the Atmel chip are the stepper motor drivers, and the chip on the left is the USB / Serial communication chip. It's says CH330 so you can use the CH340 driver.

robywoodwork, 2020-09-16:
it looks like I have a good chance.  I don't understand what it means to flash the firmware, I would be grateful if you could explain me the step by step procedure to load the firmware in the right way.  in this I am a sucker.  meanwhile, thank you for your help, you are very kind.  have a nice day

StuartB4, 2020-09-16:
Ok follow these images step by step. First Start LaserGRBL but DONT  click Connect.
If you want to change it to version 1.1f just change it in the drop down menu. 

[image]

[image]

[image]

[image]

[image]

[image]

[image]

robywoodwork, 2020-09-16:
great, you are very kind.  I'll try tomorrow and let you know.  I hope everything goes for the best.  now I am very tired, I have recently finished work.  I wish you a good evening.  thank you so much again🙏

StuartB4, 2020-09-16:
Ok. :+1:

robywoodwork, 2020-09-17:
I Send a video in mail because here no possible

robywoodwork, 2020-09-17:
I sigh

StuartB4, 2020-09-17:
You can send a video here if you want to. This is my secure email address.
stuartbassett1@protonmail.com

StuartB4, 2020-09-18:
Just been looking at the videos.
Try installing the CH340 driver on the LaserGRBL Tools menu.
Then try to Flash it again using the method on the Tools menu.
Nano, 1.1h original, 57600

[image]

[image]

robywoodwork, 2020-09-19:
you left me?

> Il giorno 18 set 2020, alle ore 16:08, Roberto M. <robywoodwork@gmail.com> ha scritto:
> 
> ﻿
> <Video.mov>
> 
> 
>> Il giorno 18 set 2020, alle ore 16:02, Roberto M. <robywoodwork@gmail.com> ha scritto:
>> 
>> ﻿I've tried them all.  the only one that uploaded me is 1.1f, and even if I try to upload its firmware it won't upload it.  it's stuck at version 1.1f which doesn't work well.  I send you the attached videos
>> <Video.mov>
>> 
>> 
>>>> Il giorno 18 set 2020, alle ore 14:50, Stuart <notifications@github.com> ha scritto:
>>> 
>>> Just been looking at the vid

StuartB4, 2020-09-19:
What do you mean?

robywoodwork, 2020-09-19:
I think I understand the problem but I don't know how to fix it.  it is not synchronized and therefore cannot write firmware.  from how I understand the causes could be different.  the card is not right, the door is not the open one.  I do not know what to do anymore.  i found a great software called avrdudess, it's simple to use, but it didn't solve my problem.  I'd already be glad it worked with its mediocre software, but at least it works.

> Il giorno 19 set 2020, alle ore 19:53, Stuart <notifications@github.com> ha scritto:
> 
> What do you mean?

StuartB4, 2020-09-19:
Thats just a frontend addition to make it easier to use Avrdude, and thats what LaserGRBL uses to flash the firmware. It's in the LaserGRBL folder in Program Files (x86). I just tried it on my Nano. With these settings. The com port will probably be different on yours.  Then click the Program button.

[image]

StuartB4, 2020-09-19:
This one is programming the 1.1f flash. 
[image]

[image]

robywoodwork, 2020-09-19:
you did it!  I don't believe it, I've been trying all morning.  so I have to press the program button in addition to the other settings.  so not the go button?  damn, now I'm working, but as I get home I try.  I can not wait.  thanks Stuart, I'll make you a statue for your patience with me

> Il giorno 19 set 2020, alle ore 22:35, Stuart <notifications@github.com> ha scritto:
> 
> Thats just a frontend addition to make it easier to use Avrdude, and thats what LaserGRBL uses to flash the firmware. It's in the LaserGRBL folder in Program Files (x86). I just tried it on my Nano. With these settings. The com port will probably be different on yours. Then click the Program button.

StuartB4, 2020-09-19:
Good luck with it. I tried it 6 times with my Nano board, 3 times with grbl 1.1f and 3 times with 1.1h alternately with these settings and it worked every time.

[image]

StuartB4, 2020-09-19:
I just tried it with the Vigotec V 1.3 firmware with the same settings and it worked fine.
If you want the Vigotec firmware I've uploaded it to my PCloud account.
This is the link, just click the Direct Download button and click Save.
https://u.pcloud.link/publink/show?code=XZ2mM5XZCo1RxYU39UhVgNUXG4xch8bKw887

[image]

StuartB4, 2020-09-19:
One more thing. 

[image]

robywoodwork, 2020-09-20:
Hi Stuart, I came back too late last night and I'm already at work this
morning. tomorrow I'm home from work and I'll try. In case it doesn't work
I was thinking of putting my computer in your hands with a team viewer what
do you think? Have a nice day. I'm Italian and you?
this is my job:
https://www.facebook.com/394412057852811/videos/343603170384274/

https://www.facebook.com/394412057852811/videos/592633668281164/

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mail
priva di virus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

arkypita, 2020-09-20:
Non riesci a programmare grbl 1.1 perché la tua porta USB comunica con il modulo ESP32 (quello del wifi per intenderci) come dimostrato dal messaggio `ets jun 8 etc etc` che è un tipico messaggio di boot dell' ESP32 mentre grbl andrebbe programmato sull'atmel (il chippettino quadrato vicino a c26) ma se non c'è connessione diretta tra l'USB e questo chip c'è poco da fare

arkypita, 2020-09-20:
Da quello che leggo la scheda vigo prevede l'aggiornamento del firmware tramite [OTA (on the air)](https://www.google.com/search?q=arduino+ota) ovvero viene aggiornata via WiFi e non via USB. 

[image]

Sinceramente ti consiglio di lasciar perdere le modifiche che stai cercando di fare perché rischi davvero di non riuscire più ad utilizzare la scheda. Tra l'altro mi pare che quella scheda abbia già di base grbl 1.1 (versione custom di vigotec) quindi non vedo motivo.

robywoodwork, 2020-09-20:
Infatti non riesco più ad usarla dopo queste prove. Il suo software dedicato non funziona più. Vorrei ripristinarlo. Ci sarà un modo. Sono fregato?

arkypita, 2020-09-20:
Sinceramente non ne ho idea.
Cerca nel suo software se c'è qualche opzione che ti permette di riscrivere il firmware (non so, sotto settings o qualcosa del genere) eventualmente prova a contattare il produttore o a vedere se esiste un forum/pagina facebook dedicata

robywoodwork, 2020-09-20:
Ho già scritto a banggood nella speranza che mi rispondano. Stuart, non siamo riusciti a ripristinare il mio laser, ma sei stato davvero gentile ad aiutarmi. Se nel frattempo trovi qualche soluzione, te ne sarei grato. Se la trovo io ti informo, magari si può aiutare qualcun altro. Ti auguro una buona giornata un sorriso, Roberto

StuartB4, 2020-09-20:
Have you tried downloading the Wifi version of the software from Vigotec. 
When you start it and go to Settings at the top there is an option to connect to your network, select your model of machine and a Restore factory setting button. 
http://vigotec.cn/software/Vigo_Works_v3.0_B20200807.zip

robywoodwork, 2020-09-20:
yes I tried but it doesn't make me select.  perhaps because there is no update yet.  I've tried them all, but I'm not giving up.  I will do other tests also because if the Vigo firmware was there before, after all these tests, the firmware is no longer there.  in fact I can't use his software.  this means that I have somehow overwritten something

> Il giorno 20 set 2020, alle ore 14:35, Stuart <notifications@github.com> ha scritto:
> 
> Have you tried downloading the Wifi version of the software from Vigotec.
> When you start it and go to Settings at the top there is an option to connect to your network, select your model of machine and a Restore factory setting button.

robywoodwork, 2020-09-21:
Hi Stuart, I reinstalled the Vigo works software, and now it works.  I thought I had lost hope.  I would have preferred GRBL or even better lightburn.  I'll have to settle for their (poor) software.  thanks again for your availability 🙏😀

> Il giorno 20 set 2020, alle ore 14:58, Roberto M. <robywoodwork@gmail.com> ha scritto:
> 
> ﻿yes I tried but it doesn't make me select.  perhaps because there is no update yet.  I've tried them all, but I'm not giving up.  I will do other tests also because if the Vigo firmware was there before, after all these tests, the firmware is no longer there.  in fact I can't use his software.  this means that I have somehow overwritten something
> 
>> Il giorno 20 set 2020, alle ore 14:35, Stuart <notifications@github.com> ha scritto:
>> 
>> Have you tried downloading the Wifi version of the software from Vigotec.
>> When you start it and go to Settings at the top there is an option to connect to your network, select your model of machine and a Restore factory setting button.

StuartB4, 2020-09-21:
Thats good to know, at least you can still use it. Well done. :+1:

robywoodwork, 2020-09-22:
Yeahhhhh

eddy64bcd, 2020-09-29:
Bonjour, je ne sais pas si je suis au bon endroit pour exposer mon problème. J'ai acheté un laser VG-L7 avec une carte controleur Vigotec et je n'arrive pas a le faire fonctionner avec Lasergbrl, j'ai réussi à la connecter avec 57600 Baud, mais il ne fonctionne pas, pourriez vous m'aider?

robywoodwork, 2020-10-01:
J'ai le même problème et à ce jour je ne l'ai pas résolu.  le logiciel liro est tellement mauvais que j'utilise le laser eleksmaker 2.5 et non le vigotec 20w.  GRBL pour toujours.  si vous le résolvez, je vous serais reconnaissant de me donner la solution.  bonne journée

> Il giorno 30 set 2020, alle ore 01:35, eddy64bcd <notifications@github.com> ha scritto:
> 
> Bonjour, je ne sais pas si je suis au bon endroit pour exposer mon problème. J'ai acheté un laser VG-L7 avec une carte controleur Vigotec et je n'arrive pas a le faire fonctionner avec Lasergbrl, j'ai réussi à la connecter avec 57600 Baud, mais il ne fonctionne pas, pourriez vous m'aider?
>

StuartB4, 2020-10-01:
I think now the only way to get it to work with LaserGRBL would be to change the control board
for a standard grbl controller. But I don't think you would find one that will fit in the same place as the vigotec board. 
All you need is one with 1 x X, and 1 x Y axis connector and the Laser 3 Pin PWM/Power connector.
There is a 4th wire (Blue) in with the laser wires on the Vigotec board. I think it's either a temperature sensor wire or the tilt sensor
that switches the laser of if it gets too hot or if it's knocked / jolted while it's working. 

Something like this board.

[image]

StuartB4, 2020-10-01:
Details. 
Communication port: Micro USB.
Supported motor: 42 stepper motor within 2A 2-phase 4-wire.
Supported software: Support VIGOTEC software，LASERGRBL,  Candle, EleksCam T2-La-ser, Benbox, GrblController, and LiteFire, etc.)
Main purpose: For DIY laser engraver.
Power supply: DC 12V 5.5 Amps.
Support axis: XY axis.

eddy64bcd, 2020-10-01:
Bonjour
Merci de votre réponse, est ce que celle ci pourrez aller? :

Spécification:
Système de contrôle de CNC à 2 axes, système de contrôle de machine de gravure laser
Modèle: C-2Axis
Système de contrôle: GRBL
Logiciel de soutien: lasergrbl (laser)
Prise en charge du système: windows XP; 7/8/10
Moteur pas à pas: 42 moteur pas à pas (1.3A/0.3Nm/24V)
Laser: 12V5Amax (1W/2.5W/5.5W/7W/10W/15W)
Puissance d'entrée: 12V (12V seulement)
Contrôle hors ligne: prend en charge gcode, txt, tap, nc, ncc, cnc et autres formats de traitement des fichiers
Type: graveur GRBL tableau de commande
Modèle: carte 1.1 GRBL
Port: port USB

https://fr.aliexpress.com/item/4000314957696.html?spm=a2g0o.detail.1000014.45.543f4d93Y0nw7B&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.14976.190542.0&scm_id=1007.14976.190542.0&scm-url=1007.14976.190542.0&pvid=15f94ec4-42c6-486c-8431-87e03b2bbd99&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.14976.190542.0,pvid:15f94ec4-42c6-486c-8431-87e03b2bbd99,tpp_buckets:668%230%23131923%2399_668%23808%233772%23903_668%23888%233325%234_4976%230%23190542%233_4976%232711%237538%23131_4976%233104%239653%235_4976%234052%2318551%2372_4976%233141%239887%237_668%232846%238115%23837_668%232717%237567%23911_668%231000022185%231000066058%230_668%233422%2315392%23435_4452%230%23189847%230_4452%233474%2316498%23919_4452%233098%239599%2374_4452%233564%2316062%23622

StuartB4, 2020-10-01:
Oui. Celui de l'image en haut de la page semble correct. Le seul problème que j'aurais avec une carte de remplacement est si le câblage du moteur pas à pas était différent dans la fiche où il se branche sur la carte. Ils devraient être les mêmes, mais ils pourraient être différents. Mais les fils peuvent généralement être retirés de la prise et réinsérés dans le bon ordre. Il y a 2 bobines dans le moteur et 4 fils, 2 fils pour 1 bobine et 2 pour l'autre. Ils doivent juste être dans le bon ordre.

Yes. The one in the picture at the top of the page looks ok. The only concern I would have with any replacement board is if the stepper motor wiring was different in the plug where it plugs in to the board. They should be the same, but they could be different. But the wires can usually be removed from the plug and reinserted in the correct order. There are 2 coils in the motor, and 4 wires, 2 wires for 1 coil and 2 for the other. They just need to be in the correct order.

eddy64bcd, 2020-10-01:
Est ce que je vais avoir besoin de configurer lasergbrl ou la configuration par défaut ira?

robywoodwork, 2020-10-02:
I thought about it but I'm not that good at fiddling with arcuino.  anyway thank you for the support.  it's a nice 20w laser but i'm using my 2.5w eleksmaker because i use GRBL.  Thanks Stuart

> Il giorno 1 ott 2020, alle ore 13:38, Stuart <notifications@github.com> ha scritto:
> 
> Details.
> Communication port: Micro USB.
> Supported motor: 42 stepper motor within 2A 2-phase 4-wire.
> Supported software: Support VIGOTEC software，LASERGRBL, Candle, EleksCam T2-La-ser, Benbox, GrblController, and LiteFire, etc.)
> Main purpose: For DIY laser engraver.
> Power supply: DC 12V 5.5 Amps.
> Support axis: XY axis.

eddy64bcd, 2020-10-02:
Merci pour votre soutien, je vais voir si j’achète cette carte.

StuartB4, 2020-10-02:
Ok Roberto. It's a shame we are so far apart, I'm in UK.  I would like to see it working.
If you decide to try at another time please let me know how you get on with it.
:+1:

StuartB4, 2020-10-02:
Ok  eddy64bcd. :+1:

robywoodwork, 2020-10-03:
I would be happy to make it work.  as I told you GRBL I love it, it's fantastic compared to vigoworks.  we should find a way to make computer and card communicate via usb.  Arkypita mentioned something to me, but for me it is beyond my knowledge.

arkypita, 2020-10-03:
I am working on it.
Seems that Vigo has put some inconsistency in their grbl implementation in VG-L7X.

Nothing impossible to handle but difficult to undertand because I don't have a Vigo VG-L7X engraver so I cannot debug it.

arkypita, 2020-10-03:
Vigo user please test this version: https://github.com/arkypita/LaserGRBL/releases/tag/v3.7.2

eddy64bcd, 2020-10-03:
Ok je vais essayer

eddy64bcd, 2020-10-03:
Cela ne fonctionne pas, les codes sources sont a copier quelque part?

arkypita, 2020-10-03:
Based on the [feedback of a user of the LaserGRBL facebook](https://www.facebook.com/groups/486886768471991/permalink/933826350444695/) community I was able to see how Vigotech has released its latest VG-L7x machine (WiFi model) with a firmware that does not respect the [grbl specifications](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface).

It is unclear which protocol they use for communication, and there are some elusive details about how it synchronizes with the PC. Here is a log of the communication between VigoWork software and the engraver: [Capture2.zip](https://github.com/arkypita/LaserGRBL/files/5322316/Capture2.zip)

There are no "ok" message that let LaserGRBL know that the message are processed, some communication look to be encoded, a 0x88 code is sent periodically and more...

Unfortunately Vigotech does not release specifications of its products, it does not provides documentation, nor does it provide user support or a technical contact to ask for information.

For those problems unfortunately it is not possible for me to implement a communication that supports this engraver.

robywoodwork, 2020-10-04:
but if you had the Vigo laser, would you be able to understand and maybe solve the communication problem?

arkypita, 2020-10-04:
I think that with an engraver under my hand (or just the control board) i could have 99% of chance to get it working by tracing protocol and by trial and error

robywoodwork, 2020-10-04:
if you want I can bring you mine and we can spend half a day in company.  maybe the problem can be solved for my joy and that of all the owners of the Vigo

Excalibur6310, 2020-10-05:
Following... robywoodwork, I am having similar issues with my vl7x. Best of luck.

[65 more replies]

closed 2021-01-28 by arkypita