arkypita/LaserGRBL #1124: Help wanted Y-axes is not working

Bjorn56, 2020-09-16:
I have a new laser and I intended to use it for engraving.
It is a laser Diy Engraver 
 It looks like one of the axis are not working.  I tried to engrave a square and here is the result
[image]
!

Bjorn56, 2020-09-16:
it should be like this
[image]

arkypita, 2020-09-16:
1) Check electric cables
2) check belt tension (not too tight, not to soft)
3) check motor current setpoint, stepper driver usually has a trimmer to set motor current)
4) maybe a died driver?
5) check grbl config (resolution, speed, acceleration)

Do some specific test to understand what kind of problem. Try run motors without load, does they move or not? Try swap x/y cable, the problem is on y again or switch to x?

BeatSlayer, 2020-09-16:
GRBL configuration?

StuartB4, 2020-09-16:
> GRBL configuration?

Start LaserGRBL, click Connect, then click Grbl menu and click Grbl configuration. Take a screenshot or photo of the settings list and post it here.

Bjorn56, 2020-09-16:
IT sere the wheels at the laser. I have loodsened them and nog it is working. The only thing is there seems to be a lack of power becuase i can hardly see ten engravement

Verstuurd vanaf mijn iPad

Bjorn56, 2020-10-05:
Hi thanks for your advice.
I loosen the boults and now it is working ok. I have another question. The laser seems lacking power. It can't  engrave (blanc) wite wood. how can I adjust the laser so the focus point is just on the surface of the object?
 I like to hear form you
 Warm Regards,
Bjorn Vosters

StuartB4, 2020-10-05:
It's harder to engrave on white material because the laser light reflects off it and does not get absorbed in to it as much as darker material. So you need to go slower with more power.

To focus the laser place a black piece of paper on top of the workpiece, put the protective glasses on, switch the laser on low power and adjust the focus ring until you get the smallest dot possible.

Bjorn56, 2020-10-05:
Ok thanks I will give it a try

closed 2020-12-16 by arkypita