arkypita/LaserGRBL #1125: Zapytanie

Belfer79, 2020-09-18:
Dzieki waszym instrukcjam udało mi sie skonfigurowac laser grbl ma jednak lekki problem po wypaleniu obrazka jego boczne krawedzie są jakby bardziej przypalone Ttak jakby laser przy zmianie kierunku  bardziej przypalal jak to moge zniwelowac

StuartB4, 2020-09-18:
Czy używasz M4 w ustawieniach obrazu docelowego. Jeśli nie, zmień go na M4 za pomocą menu rozwijanego.

[image]

Belfer79, 2020-09-21:
Dziekuje pomogło

Belfer79, 2020-09-21:
A z tą opcja linia środkowa to trzeba coś dodatkowego zainstalować  lub skonfigurować

StuartB4, 2020-09-21:
Przepraszam, nie rozumiem, co masz na myśli przez środkową linię?

Belfer79, 2020-09-22:
Chodzi o tą czwartą opcję l
[image]
Do tej opcji linia srodkowa co jest potrzebne zeby to działało bo teraz wyskakuje błąd (0×d8)

Belfer79, 2020-09-22:
Prosze mi wybaczyc za pisownie ale program sam mi słowa zmienia

closed 2020-12-16 by arkypita