arkypita/LaserGRBL #1126: SVGs from FlatCAM are imported incorrectly

positron96, 2020-09-19:
Hello! I'm trying to laser burn a PCB layout prepared in FlatCAM, exported to SVG and imported into LaserGRBL (v3.7.1). While the PCB starts at 0,0 in FlatCAM (and in Inkscape if I open the SVG in it), in LaserGrbl it is offsetted by random amount along Y axis. See pictures:

flatcam: 
[image]

inkscape: [image]

LaserGrbl  v3.7.1 [image]

Could you dig into the issue?

I'm also attaching the SVG, saved as TXT
[Gerber_BottomLayer.GBL_0.1000_iso_svg.svg.txt](https://github.com/arkypita/LaserGRBL/files/5250061/Gerber_BottomLayer.GBL_0.1000_iso_svg.svg.txt)

BeatSlayer, 2020-09-21:
I would bet that the scale operation it's the one causing troubles...

Can you try to do some operation in inkscape, like a simple ungroup, save the file and try it out?

arkypita, 2020-09-24:
[image]

The issue is with the viewbox. 
If I copy-paste the drawing to a new inkscape file, and shape out paper size to drawing size all is handled well by lasergrbl.

[image]

Look like that LaserGRBL is not able to handle viewbox correctly.
It's hard to fix because SVG routine is not a code written by me but copied from a public project and I have no idea of how to put hands on it.

positron96, 2020-09-24:
Thank you for info! 
Anyway, I've figured out that FlatCAM has a GRBL_Laser g-code preprocessor that eliminates the need for LaserGRBL's ability to import SVGs.

arkypita, 2020-09-24:
Good