arkypita/LaserGRBL #1127: M3 and M4 command

barnwellm, 2020-09-20:
Im running the latest version of LaserGbrl and have found that it will keep going back to the M3 command when I get ready to burn a project.  I have to keep going into the configuration and turning on the $32=1 but when I go back to do the burn and get it set up and set the size of the burn, it will have the M3 listed instead of the M4 command.  If you select the selection box only the M3 command is listed, and if you go back into the configuration it has the laser turned off again.  I can turn the laser back on with the $32=1 and write it to the configuration and go back and try to do the burn the M3 command is again listed.  Is there anyone else having this problem and if so is there a way to correct it?????  Any help on this would be greatly appreciated.  One more thing, this started doing the with the 3.70 version and is still doing it with the 3.71 version.  Thank you in advance

StuartB4, 2020-09-20:
It may depend on the machine you have and the controller board and if it has a custom version of grbl on the controller board. Is it a well know brand?

barnwellm, 2020-09-20:
Its just one of 3018 Pro version....the small desktop model.  Its running grbl 1.1f and has been working good until I updated the lasergrbl software.  That is when I noticed the when I loaded a file in that it would for some reason turn the M4 off and turn the M3 on when I got ready to size and burn.  I have just been playing around with it and installed a couple of custom buttons...one is to center the laser on the image.  It is for some reason now holding the M4 command.  I did have to change the code in the centering button to M3 to turn on the laser to center the laser and then issue the M4 command and then the M5 to turn the laser off...seems to be working now.  I just didnt ever have this problem until the new version of lasergrbl software.  It also is doing it in the latest update to the 3.71.  It never did it with the older version of the software.  I hope all of this makes a little sense as I am a fairly new user....so please bear with me.....:)

BeatSlayer, 2020-09-20:
PWM support activate in the LaserGRBL settings?

barnwellm, 2020-09-20:
Yes it sure is.....I have just tried it a few times in the past few minutes and for some reason it is now keeping the M4 command....Im really baffled at what was causing it and now seems to be working like it should....I know it has to be the nut behind the keyboard......!!!!  Thank you all for jumping in and trying to help....Ill play with it and see what happens and if it does it again will get back on and see if there is some suggestion out there as to what I am doing to cause the problem...again, thank you.....

arkypita, 2020-09-20:
Lasergrbl does not change $32 values by himself, as weel it does not change any $ config value by himself.
Maybe you have some custom button that change that value? Check custom button code.

barnwellm, 2020-09-20:
I did check the code in the buttons and I did find the “find center” button that would change the code…..I think now I have found and corrected the problem…..I just want to say that I tried out the “lightburn” software and will say that your lasergrbl is so much better and easier to use…thank you for all the hard work you have put into it.  The lightburn is to complicated for what you want to do….and lasergbrl does things so much better…..thank you again

Mike Barnwell
barnwellm@comcast.net

From: arkypita
Sent: Sunday, September 20, 2020 6:14 PM
To: arkypita/LaserGRBL
Cc: barnwellm; Author
Subject: Re: [arkypita/LaserGRBL] M3 and M4 command (#1127)

Lasergrbl does not change $32 values by himself, as weel it does not change any $ config value by himself.
Maybe you have some custom button that change that value? Check custom button code.

ThomasSc, 2020-09-21:
I know the problem.
It is the button "Laser on" at active.
"Laser on" active is set $32=0 M3 S2 
and the is the problem you can't set M4.

arkypita, 2020-09-21:
GRBL does not turn on laser with M3/M4 "by design" if no movement when laser mode is active, that is why your custom button set laser mode off when need (for example) to turn on led for focusing.

I have already found a better option than disable laser mode, read this post: https://github.com/arkypita/LaserGRBL/issues/306#issuecomment-653736075

closed 2020-09-21 by arkypita