arkypita/LaserGRBL #1128: Auto Cooling won't work

heathercv, 2020-09-23:
I know it is compatible because several other people with the exact same machine are using the feature. I have tried setting it at 5 minutes and 10 minutes to cool for 1-2 minutes but nothing works. I have even tried uninstalling and deleting and install and root files and reinstalling with the latest version and nothing is working. I have tried both USB and USB 2 connection ports on my computer. I am out of ideas. Running Windows if that matters and everything is definitely up to date.

BeatSlayer, 2020-09-23:
If people with the exact same machine say it's working, I don't understand why you made a bug report saying that auto cooling won't work without minding to include at least some details about the machine and configuration...

heathercv, 2020-09-23:
I have the Yora 6550 15W machine, I do not know what configuration
information you would like, but everything is set to default. I reported my
issue exactly that same way that everyone else has been doing on this
forum. I searched and read through issues before posting and followed the
same as everyone else. If there is a set form of rules or specific
information you would like included for all issues then it should be listed
as such on the main page so this issue does not come up again. Please let
me know what information you would like.

BeatSlayer, 2020-09-23:
You're right that a lot of people resort to github bug report system as a support forum, but it really isn't.

GRBL version and configuration would help diagnose it.

heathercv, 2020-09-23:
That is because the website says to come here for help

heathercv, 2020-09-23:
I installed the latest version after completely removing it and reinstalled
it fresh, I deleted all files and logs because a regular install did not
work. I have changed nothing in the settings except the auto cool feature
here is what it has for the configuration

  $0 Step pulse time 10 microseconds Sets time length per step. Minimum
3usec.
  $1 Step idle delay 25 milliseconds Sets a short hold delay when stopping
to let dynamics settle before disabling steppers. Value 255 keeps motors
enabled with no delay.
  $2 Step pulse invert 0 mask Inverts the step signal. Set axis bit to
invert (00000ZYX).
  $3 Step direction invert 2 mask Inverts the direction signal. Set axis
bit to invert (00000ZYX).
  $4 Invert step enable pin 0 boolean Inverts the stepper driver enable pin
signal.
  $5 Invert limit pins 0 boolean Inverts the all of the limit input pins.
  $6 Invert probe pin 0 boolean Inverts the probe input pin signal.
  $10 Status report options 1 mask Alters data included in status reports.
  $11 Junction deviation 0.010 millimeters Sets how fast Grbl travels
through consecutive motions. Lower value slows it down.
  $12 Arc tolerance 0.002 millimeters Sets the G2 and G3 arc tracing
accuracy based on radial error. Beware: A very small value may effect
performance.
  $13 Report in inches 0 boolean Enables inch units when returning any
position and rate value that is not a settings value.
  $20 Soft limits enable 0 boolean Enables soft limits checks within
machine travel and sets alarm when exceeded. Requires homing.
  $21 Hard limits enable 0 boolean Enables hard limits. Immediately halts
motion and throws an alarm when switch is triggered.
  $22 Homing cycle enable 0 boolean Enables homing cycle. Requires limit
switches on all axes.
  $23 Homing direction invert 0 mask Homing searches for a switch in the
positive direction. Set axis bit (00000ZYX) to search in negative direction.
  $24 Homing locate feed rate 25.000 mm/min Feed rate to slowly engage
limit switch to determine its location accurately.
  $25 Homing search seek rate 500.000 mm/min Seek rate to quickly find the
limit switch before the slower locating phase.
  $26 Homing switch debounce delay 250 milliseconds Sets a short delay
between phases of homing cycle to let a switch debounce.
  $27 Homing switch pull-off distance 1.000 millimeters Retract distance
after triggering switch to disengage it. Homing will fail if switch isn't
cleared.
  $30 Maximum spindle speed 1000 RPM Maximum spindle speed. Sets PWM to
100% duty cycle.
  $31 Minimum spindle speed 0 RPM Minimum spindle speed. Sets PWM to 0.4%
or lowest duty cycle.
  $32 Laser-mode enable 0 boolean Enables laser mode. Consecutive G1/2/3
commands will not halt when spindle speed is changed.
  $100 X-axis travel resolution 80.000 step/mm X-axis travel resolution in
steps per millimeter.
  $101 Y-axis travel resolution 80.000 step/mm Y-axis travel resolution in
steps per millimeter.
  $102 Z-axis travel resolution 800.000 step/mm Z-axis travel resolution in
steps per millimeter.
  $110 X-axis maximum rate 10000.000 mm/min X-axis maximum rate. Used as G0
rapid rate.
  $111 Y-axis maximum rate 10000.000 mm/min Y-axis maximum rate. Used as G0
rapid rate.
  $112 Z-axis maximum rate 10000.000 mm/min Z-axis maximum rate. Used as G0
rapid rate.
  $120 X-axis acceleration 500.000 mm/sec^2 X-axis acceleration. Used for
motion planning to not exceed motor torque and lose steps.
  $121 Y-axis acceleration 500.000 mm/sec^2 Y-axis acceleration. Used for
motion planning to not exceed motor torque and lose steps.
  $122 Z-axis acceleration 500.000 mm/sec^2 Z-axis acceleration. Used for
motion planning to not exceed motor torque and lose steps.
  $130 X-axis maximum travel 200.000 millimeters Maximum X-axis travel
distance from homing switch. Determines valid machine space for soft-limits
and homing search distances.
  $131 Y-axis maximum travel 200.000 millimeters Maximum Y-axis travel
distance from homing switch. Determines valid machine space for soft-limits
and homing search distances.
  $132 Z-axis maximum travel 200.000 millimeters Maximum Z-axis travel
distance from homing switch. Determines valid machine space for soft-limits
and homing search distances.

arkypita, 2020-09-23:
Turn $32=1. If you have $32=0 laser does not turn off when a pause command is sent.

$32=0 is for the usage of grbl with a CNC
$32=1 is for the usage of grbl with Laser

heathercv, 2020-09-23:
Awesome, thank you so much! I truly appreciate it!

arkypita, 2020-09-23:
You are welcome. But sometime read before ask will give you response :-)

[image]

arkypita, 2020-09-23:
Maybe is not your fault if you have $32=0
You can have received a misconfigured board from Yora, or you can have imported some custom-button that turn $32 off.

heathercv, 2020-09-23:
You litterly have the worst customer service ever lol I bought a machine
and this is what came with it, I don't program or know what any of this
stuff means at all. I was able to figure out how to change it because it's
pretty simple, but my manual says nothing about changing anything and that
isn't my fault. Your litterly paid to prove customer service and support,
if you hate doing your job that much then get a different one but don't
take it out on me. Maybe complain to Yora to provide better support but
don't take it out on the customer. Beyond unnecessary. Reading that screen
does not explain how to change anything or that it applies to my machine or
anything. It needs to be worded better and there needs to be something on
the website to explain this stuff. People don't buy machines expecting to
become programmers lol

BeatSlayer, 2020-09-23:
Maybe you didn't realize that you're a user (free), not a costumer ($$).

arkypita, 2020-09-23:
I am convinced that I am the wrong person to direct your frustration.
I am the developer of LaserGRBL, a single person that devotes his free time to create a FREE and SIMPLE program. 

I did not receive 1 cent from Yora, you have not paid 1 cent for this program and you have not paid 1 cent for this assistance, which you are receiving for FREE thanks to the time I dedicate to you, including replying to this annoying message.

I hope by now you have understood that if you are having trouble with customer service for something you paid for (hardware) you should address your complaints to whoever sold you the product, not to me.

However, if lasergrbl and its free community support are not to your liking, you are completely free to uninstall it and use another type of program.

good luck

heathercv, 2020-09-23:
And maybe you don't realize that if you actually want people to donate like
your site says then you have to make them want to. Perhaps you need to
consider making a contract with Yora, I don't know, but I don't deserve the
uncalled for attitude. It isn't my fault your where I was lead to and
perhaps you should consider using a form for submission of issues to ensure
you have all the details you need up front next time. Peace

arkypita, 2020-09-23:
Thanks for your hint

StuartB4, 2020-09-23:
Wow. Was not expecting that.

closed 2020-12-16 by arkypita