arkypita/LaserGRBL #1130: habe das selbe problem auch mit m4 gibts eine lösung

fude57, 2020-09-24:


BeatSlayer, 2020-09-27:
Be sure to enable "laser mode" by setting $32=1 in the GRBL configuration.

Danotec, 2020-11-02:
Wenn Sie Einbrand an den Start und Endpunkten im Bild haben, versuchen Sie die Parameter $120 und $121 ( X und Y Beschleunigung der Motoren zu erhöhen.
Ich denke mit Werten von 1800-2500 mm/sek²  sollte das Problem gelöst sein.
________________________________________________
If you have brand at the start and end points in the picture, try increasing the parameters $ 120 and $ 121 (X and Y acceleration of the motors.
I think with values of 1800-2500 mm / sec² the problem should be solved.

Grüße!

fude57, 2020-11-02:
Besten dank, werde ich ausprobieren
G rolf

Gesendet mit der blue News & E-Mail App

Danotec, 2020-11-02:
Achten Sie darauf das Ihre Motoren und auch  die Motorhalter sowie Antriebselemente die Belastung auch auf dauer aushalten können, ggf. besser mit den Beschleunigungswerten etwas zu wenig als zu viel.

__________________________________________________________________________________________

Make sure that your motors and also the motor mounts as well as drive elements can withstand the load in the long term, if necessary with the acceleration values a little too little than too much.

fude57, 2020-11-03:
Habe von 110mm sec auf 300 mm gestellt und problem ist gelöst, besten dank für tip 

Gesendet mit der blue News & E-Mail App

closed 2020-12-16 by arkypita