arkypita/LaserGRBL #1131: Laser suddenly no longer engrave/cut

edoardoado, 2020-09-24:
Hi, i have a Alfawise C30 Pro 3000mw 445nm 12V by 4 months.

It was working perfectly, but it suddenly stop working properly:
- it doesn't switch on with the command M3S1 (only from S20 or more) and it stay on for few seconds only.
- i can set any speed/power but it will not engrave/cut anything at all (wood/paper doesn't matter)

I didn't upgrade the computer or the software, it just looks like the laser lost power (the lens is clean), software/connection/motors work fine

Thanks

StuartB4, 2020-09-24:
Sounds like the laser may be overheating and switching off. 
Does the engraver have this control board? 

[image]

edoardoado, 2020-09-24:
No it use this kind

[image]

StuartB4, 2020-09-24:
Oh ok. Sorry I don't know enough about that board. 
Just curious though, where do the 3 black wires plugged in to the Probe connector go to?

edoardoado, 2020-09-25:
That's a shock sensor to stop the machine in case it get hit

StuartB4, 2020-09-25:
Oh ok, thanks.

closed 2020-12-16 by arkypita