arkypita/LaserGRBL #1132: New Laser Master 2 20w will not engrave

jayronayne, 2020-09-27:
I cannot get my new Ortur Laser Master 2 to engrave using Laser GRBL.  When I hit the "laser" custom button it fires a burn mark into my work piece.  When I attempt to engrave, it seems to run correctly but it doesn't seem to have any power from the laser so it doesn't burn anything at all. 

 What am I missing?  Any help is appreciated.

BeatSlayer, 2020-09-29:
Does the laser turn on?
Which speed and power did you selected?

MaiLes839, 2020-10-03:
Hi, I have the same problem. the laser head is going through the motions but doesn't burn the wood. 
There is a laser blue light so I'm assuming the laser is turned on, I tried all sort of power and speed but now I tried the highest one and it still doesn't work
[image]

I will appreciate your help, I and struggling with this and can't find the answer anywhere online... thanks!

gmmanonymus111, 2020-10-03:
I think (by the picture you uploaded) you are setting the laser power at a wrong place. You should set it back to 1x.
You can change laser power during importing an image/vector (I was importing a PNG image):
[image]
Alternatively, if you want to test it, you can use the following commands

- If you have lasermode enabled (`$32=1` in GRBL->GRBL configuration):
`G1 F1 M3 S*` where * can be any number between 1-255 (or your laser maximum, configured at `$30` (Maximum spindle speed). Set it to 0 to turn your laser off.

- If you have no lasermode (`$32=0`):
`M3 S*` should be enough, the formula is the same as above ^

MaiLes839, 2020-10-03:
Thank you so much for answering me.
I still can't make it work though, I tried changing the power of the laser where you showed in the image, but still nothing... and 

[image]

I didn't quite understand your last comment...
I only have M3 (no M4), and I have no lasermode ($32=0):

[image]

Where do I need to put the formula you wrote on your comment: G1 F1 M3 S* (1000 - this is the number that is in $30).
Do you have any other thoughts on what do I do wrong...?

Thanks!

gmmanonymus111, 2020-10-03:
You need to type GCode commands here once you are connected to your machine:
[image]
According to your pictures, the command should be: `G1F1M3S5000` for 50% power (as max power is set at 10000, not 1000). This should light up your laser

Also, if you intend to engrave with a laser, you should enable laser mode (it'll give you finer images, and when your machine is on pause, GRBL will turn off the laser automatically for safety reasons)

MaiLes839, 2020-10-03:
OK I tried that - changed to enable laser mode and added the code but still nothing has changed....

[image]

[image]

gmmanonymus111, 2020-10-03:
It looks like the laser is out of focus, the blue dot is too big. Try to rotate the lens/increase or decrease the height of the material you try to engrave, if the laser is a fixed focus one. If it is already focused, I'm afraid the diode is dead.

MaiLes839, 2020-10-03:
you are an angel, but still - I tried everything you said and nothing is happening ):

gmmanonymus111, 2020-10-03:
As a last resort, you should try engraving a grayscale image, like this one:
[image]
The settings should be the following:
[image]
[image]
This should take ~ 15 mins, but it will go from 0 to 10000, so you can check where (if) it starts to burn the wood.
What model and how old is the engraver by the way?

MaiLes839, 2020-10-03:
It worked!!!!! can you explain maybe why? so I will understand what I did wrong...?
(it engraved it as a mirror writing though - how come...?)
I just bought the engraver, this is the first time I tried it.... it is a mini CNC 20W 7000 MW STUMP
Thanks so much!!

gmmanonymus111, 2020-10-03:
Well, I think it needs an effective power range and speed. You need to check where it started burning (it is a 5.5W laser according to the online description, just like mine, so maybe around ~3000 @ 600mm/min). This will be really time consuming, because you have to try a lot of settings. Luckily, LaserGRBL has a material DB (`GRBL`->`Material DB` or click on the book icon when importing an image), which can help you set the correct values.
[image]

As for the mirror writing, you might need to change the direction of one axis. (use `$3` at `GRBL`->`GRBL Configuration`, `$2` is just for reference)
[image]
More info of the configuration can be found [here](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration)

MaiLes839, 2020-10-03:
Thanks again for all your explanation, I really appreciate your help!!!!! It was super useful (:

gmmanonymus111, 2020-10-03:
Glad I could help ^^

MaiLes839, 2020-10-12:
Alex, do you know what can be the reason for these (unwanted) lines being engraved and how I might be able to avoid them? Thanks
[image]

StuartB4, 2020-10-12:
Have you tried increasing the White clip slider where you prepare the image.
[image]

MaiLes839, 2020-10-12:
I will give it a try, thanks!

closed 2020-12-16 by arkypita