arkypita/LaserGRBL #1133: Newbie help: Laser will not home

Riftburn, 2020-10-03:
Good morning. I recently purchased the Ortur Laser Master 2 7w. I have looked for trouble shooting help for the 7W but every video I find is the 15 or 20. Here is my issues

I put together the equipment, connected to my computer, ensured I had power. After setting up in both Lightburn and in Laser GBRL I attempted to do a test burn. My laser will move on the X axis but not the Y. When I select home the motor spins and then proceeds to go max to the right on the Y axis until it reaches the end and tries to continue until it error codes out. I tried to find additional videos to ensure I completed my set up right but honestly, i am clueless. I have included a picture of the console box.

Also, please, talk to me Barney style, like I am 5 years old. lol
[image]

gmmanonymus111, 2020-10-03:
I think the 7W, 15W and 20W engravers are the same, the only difference is the laser module, so the assembly should be almost the same.
You should make sure the wiring is correct (it sounds like your limit switches and a stepper motor are not connected correctly.

Riftburn, 2020-10-03:
Thank you, I will try that. The instruction video from Ortur shows the 15W
which has different cables it looked like.

gmmanonymus111, 2020-10-03:
I might have [found a video](https://www.youtube.com/watch?v=etpRwRNtS78) that could help you (I don't know if it is the same machine as yours, but he had a similar issue with limit switches). This video also shows a complete assembly, yours should be similar.

gw935, 2020-10-15:
I habe the same problem. I also purchest the Ortur Laser Master 2 7W and exactly the same thing happens.
The only difference is that, with LaserGRBL I am able to move the Laser.
When I move the Y-Axis everything is alright but when I attemt to move the X-Axis to the right, 
the laser moves to the left and vice versa.

arkypita, 2020-10-16:
7/15/20W it makes no difference

@Riftburn  Check the cable. I see a lot of people inverting cables connector.

[image]

@gw935 you can reverse axis direction in grbl configuration. I have written a document http://lasergrbl.com/configuration/

NOTE: This is not the Ortur support page. If you need support for Ortur engraver please ask to their community/forum
https://www.facebook.com/groups/orturusergroup
https://ortur.tech/support/

closed 2020-12-16 by arkypita