arkypita/LaserGRBL #1134: Commands G2 and G3 are not displayed correctly

Vasilius-001, 2020-10-03:
There is a laser engraver with GRBL 1.1h firmware. I need to draw circles and arcs, for this I create in a manual file. I open the file in LaserGRBL 3.7.1. Circles are displayed in the preview, but arcs are not displayed at all. In this case, the image with a laser engraver is correct.
```
G90 (use absolute coordinates)
M5
G00 X300 Y300 F2000
M3 S80

(Circles R150 - R148)
G00 X0.000 Y150.000 F2000 (circles R150.000)
G02 X0.000 Y150.000 I150.000 J0.000 F2000
G00 X0.125 Y150.000 F2000 (circles R149.875)
G02 X0.125 Y150.000 I149.875 J0.000 F2000
G00 X0.250 Y150.000 F2000 (circles R149.750)
G02 X0.250 Y150.000 I149.750 J0.000 F2000
G00 X0.375 Y150.000 F2000 (circles R149.625)
G02 X0.375 Y150.000 I149.625 J0.000 F2000
G00 X0.500 Y150.000 F2000 (circles R149.500)
G02 X0.500 Y150.000 I149.500 J0.000 F2000
G00 X0.625 Y150.000 F2000 (circles R149.375)
G02 X0.625 Y150.000 I149.375 J0.000 F2000
G00 X0.750 Y150.000 F2000 (circles R149.250)
G02 X0.750 Y150.000 I149.250 J0.000 F2000
G00 X0.875 Y150.000 F2000 (circles R149.125)
G02 X0.875 Y150.000 I149.125 J0.000 F2000

(Arc R147.875 - R147)
G00 X2.125 Y150.000 F2000
G02 X297.875 Y150.000 R147.875 F2000 (arc R147.875)
G00 X297.750 Y150.000 F2000
G03 X2.250 Y150.000 R147.750 F2000 (arc R147.750)
G00 X2.375 Y150.000 F2000
G02 X297.625 Y150.000 R147.625 F2000 (arc R147.625)
G00 X297.500 Y150.000 F2000
G03 X2.500 Y150.000 R147.500 F2000 (arc R147.500)
G00 X2.625 Y150.000 F2000
G02 X297.375 Y150.000 R147.375 F2000 (arc R147.375)
G00 X297.250 Y150 F2000
G03 X2.750 Y150.000 R147.250 F2000 (arc R147.250)
G00 X2.875 Y150 F2000
G02 X297.125 Y150.000 R147.125 F2000 (arc R147.125)
G00 X297.000 Y150.000 F2000
G03 X3.000 Y0.000 R147.000 F2000 (arc R147.000)

M5
G90
G00 X0 Y0 F2000
M30
```

arkypita, 2020-10-04:
I have implemented math only for I-J format of G2/G3.
I guess it could be simple to implement, but I am not so good in math :-(
```

public G2G3Helper(LaserGRBL.GrblCommand.StatePositionBuilder spb, LaserGRBL.GrblCommand cmd)
{
	bool jb = cmd.JustBuilt;
	if (!jb) cmd.BuildHelper();

	double aX = (double)spb.X.Previous; //startX
	double aY = (double)spb.Y.Previous; //startY
	double bX = (double)spb.X.Number;	//endX
	double bY = (double)spb.Y.Number;	//endY

	double oX = cmd.I != null ? (double)cmd.I.Number : 0.0; //offsetX
	double oY = cmd.J != null ? (double)cmd.J.Number : 0.0; //offsetY

	CenterX = aX + oX; //centerX
	CenterY = aY + oY; //centerY

	Ray = Math.Sqrt(oX * oX + oY * oY);  //raggio
	RectX = CenterX - Ray;
	RectY = CenterY - Ray;
	RectW = 2 * Ray;
	RectH = 2 * Ray;

	StartAngle = CalculateAngle(CenterX, CenterY, aX, aY); //angolo iniziale
	EndAngle = CalculateAngle(CenterX, CenterY, bX, bY); //angolo finale
	AngularWidth = AngularDistance(StartAngle, EndAngle, spb.G2);

	if (!jb) cmd.DeleteHelper();
}
```

gerdvogel, 2021-01-04:
Hello,
I found this something to convert the G02/G03 problem with radius to center coordinates.
If you assume that you put a circle with radius R at the start and end point of the arc and where the two circles touch is the center of the arc, you can convert these coordinates to I and J.
The page is unfortunately in German, has a JavaScript to calculate the center. Maybe you can have a look at it and include it in your Conde. Unfortunately I don't have a compiler on my computer to help you with this. 
I just found out that Microsoft Edge can also translate web pages. I personally use Deepl as a translator, it's online.

http://walter.bislins.ch/blog/index.asp?page=Schnittpunkte+zweier+Kreise+berechnen+%28JavaScript%29