arkypita/LaserGRBL #1135: antivirus blocks download

metzindustries, 2020-10-04:
latest version 3.7.1 is showing positive for trojan virus by 360 anti virus

Previous version  3.7.0 no problem.

Also with windows 7 no highlighted pointer showing tool location on the work job.

arkypita, 2020-10-04:
False positive: http://lasergrbl.com/faq/

Scanned with VirusTotal (test against 70 AV engine) only one (false) positive.
https://www.virustotal.com/gui/file/ec2de7af69f0bbea8abcd236a7438d8d2c81dd78de1d07521a276b28fa5b4072/detection

Mrjavaci, 2021-01-21:
I guess it's blocking the download because you don't have an SSL certificate.

For free ssl, I recommend using CloudFlare.

If you don't know how, you can contact me about the subject.

iletisim@osmantahakiyicilar.com.tr