arkypita/LaserGRBL #1137: Czwarta opcja w lasergrbl

Belfer79, 2020-10-04:
[image]
 Mam pytanie co trzeba zrobić żeby tą opcja czwartą była aktywna

StuartB4, 2020-10-04:
Z Vectorize robi to.

[image]

Z linią środkową wykonuje po prostu 1 linię w oparciu o środek takiego obiektu.

[image]

[image]

[image]

arkypita, 2020-10-04:
Centerline need 64bit OS

arkypita, 2020-10-04:
http://lasergrbl.com/faq/

closed 2020-10-04 by arkypita