arkypita/LaserGRBL #1138: Cycles in custom buttons?

ssombra, 2020-10-09:
is there a way to make a push button that makes the laser go around the image for as long as the mouse button is pressed? 
Some kind of "while" cycle.

ssombra, 2020-10-09:
I guess a 2 state button could also work for that purpose. The idea is framing the image more than one pass.

StuartB4, 2020-10-10:
You can edit the Custom buttons.
This is the Frame button code, it's just repeated 4 times so it will frame 4 times.
You could repeat it as many times as you want, just by adding the same code over and over.
To change the speed just increase or decrease the F numbers in the code.

[image]

arkypita, 2020-10-10:
Hi @ssombra 
It is not possible. Grbl (firmware) doesn't support gcode for cycles and implement them in LaserGRBL software it is not simple because you cannot know when one cycle execution is ended and you can send next) because Grbl give feedback only of command received, not executed.

ssombra, 2020-10-11:
Thanks both for the explanations.

closed 2020-10-11 by ssombra