arkypita/LaserGRBL #1139: The laser leaves marks (engraved lines) when traveling back and forth

MaiLes839, 2020-10-10:
Hi, when I engrave an image like the below, the laser engrave unwanted lines when - how can I avoid this?
Thanks

[image]

StuartB4, 2020-10-10:
Check grbl configuration on the Grbl menu. Look for $32 in the list and make sure it's set to 1.
Set $30 to 255.
Then click Write and Close. 

When you prepare your image with the brightness, contrast, etc. 
Click Next and make sure Laser ON is set to M4.

MaiLes839, 2020-10-12:
thanks, but unfortunately it didn't work, still engraving these lines...

BeatSlayer, 2020-10-12:
Do they appear on the preview?
It doesn't really look like engraving while traveling because there are also some lines missing, could it be a connection issue?
Are you connecting at the recommended bitrate for your board?
Are you using a thick shielded usb cable away from the motor wires?

MaiLes839, 2020-10-13:
Hi, here is the preview and they do not appear there (unless there is something I am missing):
[image]

I'm not sure I understand your question: "Are you connecting at the recommended bitrate for your board?" - where can I configure this?

Thanks!

gmmanonymus111, 2020-10-13:
I think @BeatSlayer meant the baud rate (aka bitrate) of your connection. It can be set before connecting to your machine
[image]

But, as your machine connects (and LaserGRBL receives status messages), I would say the bitrate is fine.
It seems your engraver does not receive some of the commands sent by LaserGRBL. I would try with a different cable (most wired printers use the same kind, if you have one, you could try with that, they usually better in quality than chinese ones bundled with arduinos/engravers).
You may also try to lower the status report requests in LaserGRBL (`Grbl`->`Settings`), to either `Fast `or `Slow` from `UltraFast`.
[image]

MaiLes839, 2020-10-14:
Thanks Alexs!
I changed to "Slow" but still didn't work, I will try changing the cable (it is actually not from China - they didn't send any cable with the machine, but I will try anyways...)

closed 2020-12-16 by arkypita