arkypita/LaserGRBL #1140: No sustained connection between LaserGrbl and Cosmos Maker controller

jamesmknox, 2020-10-11:
System:  Windows 10 desktop.  I have a CNC 3018 with a Cosmos Maker (aka Annoy Tools) controller.  This controller has the admittedly notorious CH340G USB - Serial chip.  It works okay under UGS, although it sometimes does take multiple tries.  However, I downloaded LaserGrbl (ver 3.7.1) a couple of days ago, and am at a roadblock.  It installed fine, and seems to run fine.  However, it refuses to make a meaningful connection to the CNC controller.

Appropriate COM port and baud rate.  Clicking "connect" produces an indication that it connected - specifically I get back a message "Grbl 1.1f ("$" for help)".  The status changes to "Connecting".  However, the jog and other buttons never enables, and it will take no commands.  About 10 seconds later there is a "bing bong" and status changes back to "Disconnected".  

Log shows:
10/11/2020 12:39:47	Program     	------- LaserGRBL v3.7.1 [64bit] START -------
10/11/2020 12:40:04	SetStatus   	Machine status [Disconnected]
10/11/2020 12:40:20	SetStatus   	Machine status [Connecting]
10/11/2020 12:40:20	OpenCom     	Open COM4 @ 115200 baud 
10/11/2020 12:40:20	SendImmediate	Send Immediate Command [0x18]
10/11/2020 12:40:22	VersionInfo 	Detected Grbl v1.1f
10/11/2020 12:40:30	OpenCom     	Connection timeout!
10/11/2020 12:40:30	CloseCom    	Close COM4 [CORE]
10/11/2020 12:40:30	SetStatus   	Machine status [Disconnected]

I've tried reinstalling the device driver (which appears to be the exact same one I was using with UGS).  I've rebooted,.  I've tried running LaserGRBL with administrator privilege.  It doesn't get worse, but it doesn't get better either.  Still, I can cut and engrave with UGS... just not use the laser with LaserGRBL.  

Any ideas/suggestions appreciated.   jmk

arkypita, 2020-10-11:
LaserGRBL need to recognize valid grbl status report to accept valid connection.
status report is a message that the controller send to LaserGRBL to report its state and position and look like `<Idle|MPos:0.000,0.000,0.000|FS:0,0>`

Can you record an "extended log" (menu ? - Activate extended log) of a connection session and upload here?

jamesmknox, 2020-10-11:
Unfortunately, yes - all too easily.

00000	00155524	log	Recording session started @ 10/11/2020 14:50:34
00001	00160232	com	Open COM4 @ 115200 baud 
00002	00160242	tx	[18]
00003	00160244	tx	[3F]
00004	00160447	tx	[3F]
00005	00160647	tx	[3F]
00006	00160848	tx	[3F]
00007	00161047	tx	[3F]
00008	00161247	tx	[3F]
00009	00161449	tx	[3F]
00010	00161648	tx	[3F]
00011	00161824	rx	\r
00012	00161827	rx	Grbl 1.1f ['$' for help]\r
00013	00161848	tx	[3F]
00014	00162048	tx	[3F]
00015	00162248	tx	[3F]
00016	00162447	tx	[3F]
00017	00162648	tx	[3F]
00018	00162847	tx	[3F]
00019	00163047	tx	[3F]
00020	00163249	tx	[3F]
00021	00163447	tx	[3F]
00022	00163647	tx	[3F]
00023	00163848	tx	[3F]
00024	00164048	tx	[3F]
00025	00164248	tx	[3F]
00026	00164447	tx	[3F]
00027	00164647	tx	[3F]
00028	00164847	tx	[3F]
00029	00165047	tx	[3F]
00030	00165247	tx	[3F]
00031	00165447	tx	[3F]
00032	00165648	tx	[3F]
00033	00165848	tx	[3F]
00034	00166048	tx	[3F]
00035	00166248	tx	[3F]
00036	00166447	tx	[3F]
00037	00166647	tx	[3F]
00038	00166848	tx	[3F]
00039	00167048	tx	[3F]
00040	00167248	tx	[3F]
00041	00167447	tx	[3F]
00042	00167647	tx	[3F]
00043	00167847	tx	[3F]
00044	00168047	tx	[3F]
00045	00168247	tx	[3F]
00046	00168449	tx	[3F]
00047	00168647	tx	[3F]
00048	00168848	tx	[3F]
00049	00169048	tx	[3F]
00050	00169248	tx	[3F]
00051	00169448	tx	[3F]
00052	00169647	tx	[3F]
00053	00169848	tx	[3F]
00054	00170048	tx	[3F]
00055	00170235	com	Close COM4 [CORE]
00056	00174360	log	Recording session stopped @ 10/11/2020 14:50:52

arkypita, 2020-10-11:
Look like it does not reply at all with status message!

arkypita, 2020-10-11:
Usually comlog llok like this:

`00000 00119782	log	Recording session started @ 12/3/2019 12:14:20 AM
00001	84893492	com	Open COM21 @ 115200 baud 
00002	84893834	tx	[3F]
00003	84893936	com	Close COM21 [CORE]
00004	84896505	com	Open COM21 @ 115200 baud 
00005	84896540	com	Open COM2 @ 115200 baud 
00006	84900283	com	Open COM21 @ 115200 baud 
00007	84900371	tx	[3F]
00008	84900410	rx	<Idle|WPos:-116.619,-101.500,0.000|Bf:15,128|FS:0,0|Ov:100,100,100>\r
00009	84901391	tx	[3F]
00010	84901420	rx	<Idle|WPos:-116.619,-101.500,0.000|Bf:15,128|FS:0,0>\r
00011	84902384	tx	[3F]
00012	84902413	rx	<Idle|WPos:-116.619,-101.500,0.000|Bf:15,128|FS:0,0>\r
00013	84903391	tx	[3F]
00014	84903419	rx	<Idle|WPos:-116.619,-101.500,0.000|Bf:15,128|FS:0,0>\r
00015	84904382	tx	[3F]
00016	84904411	rx	<Idle|WPos:-116.619,-101.500,0.000|Bf:15,128|FSmo`

arkypita, 2020-10-12:
@jamesmknox do you have the offline controller mounted on it?

jamesmknox, 2020-10-12:
No, but I found the problem - thanks to your suggestion about the extended LOG.  Traced the problem down to ANOTHER broken SMD resistor.  [Second one I've found on this board, and also a previously found pulled lead on an IC.]  So, looks like I've got it working.  [At least, until I hit the next problem. <G>]  My great thanks for the help.   jmk

closed 2020-12-16 by arkypita