arkypita/LaserGRBL #1141: same image different engraving speed

ronisegoly, 2020-10-16:
Hi
I am looking for smart way to re engrave parts of imported image with different speed
I can of course erase parts of the image, making sure I keep the borders, so the SW does not change anything, but is there a smarter way?
[image]

StuartB4, 2020-10-16:
You can cut out the part you need and save it as a seperate file, then after you have loaded the main image in click append file and load it in and give it different speed and power settings.

ronisegoly, 2020-10-17:
I have played a bit, the best method for me now is when I use graphic SW which enables layers, to create basic layer with just 4 dots at each corner (to maintain size), and the then export to different images different  parts of the whole image, and then append on lasergrbl.
I engrave wood, so the only reason was to redo parts of the image for better results, without re engraving the whole image.
It opens new possibilities

StuartB4, 2020-10-17:
Have you tried Lightburn?

ronisegoly, 2020-10-17:
No, will look

StuartB4, 2020-10-17:
No it's a 30 day trial then it stops working unless you pay for a 12 month license. 
That costs $40 but you get updates for a year. 
Then if you don't want any more updates after the first year it keeps working as normal.
If you pay for another year it only costs $30 and you get the updates again.
I just payed for a second year and got 2 months extra free. So the new license is for 14 months. 
If you are going to use it a lot it is well worth it.

StuartB4, 2020-10-17:
Also you can use the license on 2 computers, but if you need it on 3 just contact them and they will usually allow you to do that.

ronisegoly, 2020-10-17:
I looked briefly.
What does it adds to what I have today? Able to import images to lasergrbl, and I have all the flexibility with graphics using my own tools

StuartB4, 2020-10-17:
Too much for me to go in to detail here. If you have the time, ahve a look at some of the video tutorials, they will give you a better idea than i could here.
https://lightburnsoftware.com/pages/tutorials

closed 2020-12-16 by arkypita