arkypita/LaserGRBL #1143: Need Help Assembling Laser Module, CNC Shield amd Programming

aceride23, 2020-10-17:
[image]
[image]
[image]
[image]
[image]

Hi all, I bought a laser cutter kit off Banggood, but the MKS board was defective so instead of buying another one I decided to opt for a CNC shield v3. However, this proved to be more challenging than I initially thought. I also could not find any information similar to my configuration online. So my questions are

1. How to I connect my 3500mW laser to the Arduino uno
2. I have a dual Y, should I plug the motor in the Z or the additional axis slot?
3. What should I change on configuration.h?

Thank you in advance!

BeatSlayer, 2020-10-17:
1- one pin from the controller board for PWM, power directly from the PSU to the laser module.

2- just google "connecting stepper motors in series.

3- board type, drivers, pins, etc.

StuartB4, 2020-10-17:
Like this. On mine in this picture the black and red go to a permanent supply, it can be the same supply as the one powering the shield, the yellow and black got the the Z+ and Ground on the row of pins down the right hand side of the shield, all the pins on the far right edge are ground. 
[image]

StuartB4, 2020-10-17:
For the Y axis stepper motors one has to be connected the opposite way to the other on the shield. Because when you look at the machine working one is going clockwise and the other is counter clockwise when the shafts are facing each other.

StuartB4, 2020-10-17:
You should'nt need to change anything in config.h. Just flash the Arduino with grbl 1.1h and then change the settings in grbl configuration on the grbl menu in LaserGRBL.

StuartB4, 2020-10-17:
The heatsink on the bottom left motor driver board is really close to the edge, make sure it's not touching the pin heads underneath, it will short out if it is.

aceride23, 2020-10-18:
Hi @StuartB4 and @BeatSlayer , thanks for your pointers
[image]

I have connected the printer in the way you described. I have also flashed it with GRBL 1.1. However, currently only the X axis can move. 

Forgot to mention that I mainly use a Mac (Openbuilds), however have taken out a Windows pc to see if it would work with LaserGrbl. On laser GRBL v3.8.0 the com port wouldn't even connect. Any ideas?

Thanks again for all your help.

aceride23, 2020-10-18:
[image]

aceride23, 2020-10-18:
[image]

StuartB4, 2020-10-18:
Have you installed the driver on the PC. There's one on the Tools menu in LaserGRBL.

StuartB4, 2020-10-18:
I noticed on the image you posted at the top of this post, you have the yellow and white lead connected to the PWM socket on the laser board. 
You have the Yellow on the Ground pin on the laser board, but on the shield Z+ it's the wrong way round on the image you posted today. The yellow wire should be on the ground pin on the shield to match and White on Z+.

StuartB4, 2020-10-18:
Also the driver board you swapped to the other socket is the opposite way round to what it was before. It's upside down, that may be why the Y axis is not moving.

StuartB4, 2020-10-18:
[image]

aceride23, 2020-10-18:
[image]
[image]
[image]

Hi @StuartB4 I've swapped the stepper away und but still not luck. My laser doesn't seem to be lighting up either :/

aceride23, 2020-10-18:
[image]
 
Perfect laser working now. Thanks @StuartB4 :) just y axis left now!

StuartB4, 2020-10-18:
The little black jumpers  under the motor driver boards are for microstepping. They break up bigger steps from the motor to smaller steps, so theres more steps for 1 revolution of the motor, you get smoother movement but less torque from the motor.
As for the Y axis, does it do anything at all, or are both motors dead.

StuartB4, 2020-10-18:
Also, the little jumper on the EN/GND pins is not needed, thats for external driver boards. Take it off and just place it on the outer pin so you dont lose it.

aceride23, 2020-10-18:
Both motors aren't moving, but I just checked they are both working individually.

StuartB4, 2020-10-18:
Thats strange. 

I don't think it would make any difference to yours, but I noticed on mine the X and the bottom Y motors plugs are connected the same way. 
The top Y has the plug rotated 180 degrees.

[image]

aceride23, 2020-10-19:
> Thats strange.
> 
> I don't think it would make any difference to yours, but I noticed on mine the X and the bottom Y motors plugs are connected the same way.
> The top Y has the plug rotated 180 degrees.
> 
> [image]

Just tried to mimic yours. No luck.. Is your board Protoneer v 3 or 3.5? Mine's v3.0

StuartB4, 2020-10-19:
The only difference between your board and mine really is mine uses the SpnEn (Spindle Enable) pin for the PWM signal and yours uses the Z+ or Z-, thats because grbl changed the pin assignments from one version to another.

The other difference's are there is no glass fuse on mine, on your's it's between 2 of the stepper driver sockets on the left.  It's not needed apparently so they ditched it. 

There is resistor and capacitor filtering on mine to help prevent false triggering for limit switches.
Apart from that they both work the same. 

I've only just upgraded from the v3 to the 3.51, but I did'nt have to change or modify anything to use the existing setup I had. 
Just the PWM pin was different. So if your stepper drivers and motors and power supply are ok it should work how it is. 

Is your power supply strong enough to run the motors and the laser.
Also have you tried setting the current for the motors on the stepper driver boards. 
You have to do it with the motors disconnected from the shield. 

This is how to do it, but you will need a multimeter. 
https://www.youtube.com/watch?v=YJSmrbeZUp4

aceride23, 2020-10-20:
> https://www.youtube.com/watch?v=YJSmrbeZUp4

Thanks I'll have a look after quarantine! Left my multimeter behind.. :(

StuartB4, 2020-10-20:
Ok.

closed 2020-12-16 by arkypita