arkypita/LaserGRBL #1144: (Newbie) Please help My laser turns on when it is suppose to be off

Ahhhmyeyes, 2020-10-19:


Ahhhmyeyes, 2020-10-19:
Hello. My laser is an add on to my CNC machine

For some reason when running code, my Laser is on when going between paths and off when it is supposed to be engravening. Is there a setting to reverse this that I am not understanding?

I am running GRBl on an Uno and the laser is a Siansmart 2500mw. It states on the top of the laser that on = 0 and off = 1. Does this have something to do with it? If so can something be done?

Thanks for any help anyone can give me.

gmmanonymus111, 2020-10-19:
It seems your driver uses reversed TTL signals
Please take al look at #280

BeatSlayer, 2020-10-19:
To sum it up, you should invert the signals.

It'll cost you like $1 and you don't really need to modify the hardware if you make some small adapter board.

Ahhhmyeyes, 2020-10-20:
Grate thanks so much for the help. I will look in to that.

closed 2020-12-16 by arkypita