arkypita/LaserGRBL #1145: Laser out of the pre-defined boundries

MaiLes839, 2020-10-22:
Hi, when I'm adding a new image for engraving, I'm adding the image size, then running the "Boundary" button (see image below) to make sure the boundaries are OK. It seems to be perfect, but when I start the engraving job, the laser tries to travel beyond these boundaries (it get stuck since it is beyond the machine's engraving area. 
What am I doing wrong? something with the configuration? I gave $130 and $131 for the surface area of the machine...
Any help would be much appreciated! thanks
[image]
[image]

gmmanonymus111, 2020-10-22:
Could you try it with a smaller picture, and share the generated GCode with us, if the problem occurs? (`File`->`Quick Save`, then you can attach the file here)
When (after) it hit the limits, does limit switches shut your machine off, or does it try to move in other directions after a while?
What kind of engraver do you have?

MaiLes839, 2020-10-22:
Thanks Alex!

_When (after) it hit the limits, does limit switches shut your machine off, or does it try to move in other directions after a while?_
Neither. It tries to continue in the same direction (but of course can't. It only stops when I click the stop button manually.

_What kind of engraver do you have?_
It's called STUMP:
[image]

_EDIT: When it is stuck, could you send a screenshot of your LaserGRBL Command output window?_ 
[image]

It stops only when I click the "STOP" BUTTONbutton:

[image]

arkypita, 2020-10-22:
Look like your $100, $101, and $102 is not correctly configured.

http://lasergrbl.com/wp-content/uploads/2020/05/Grbl-Configuration-ENG.pdf

gmmanonymus111, 2020-10-22:
**“error:25” : _(“A G-code word was repeated in the block.”)**
This can be either because LaserGRBL sends something it shouldn't (I doubt it), or by electrical noise during communication (more likely). It is hard to tell without a sample GCode.
In the meantime, you can read the section "_Weird result, hang, reset or errors._" of the faq [here](https://lasergrbl.com/faq/)

StuartB4, 2020-10-22:
Looks like your settings are wrong for $130 and $131. X axis is 130. From the picture you posted above the X axis should be 200. and the Y 170

StuartB4, 2020-10-22:
Unless they do it the other way round on that machine. Using the bar that the laser slides on as the Y axis.

MaiLes839, 2020-10-22:
Alex - thanks I will try what you said.
Stuart - I think the settings are OK because when I do a test run for the boundaries before starting the job (see image below) it marks the boundaries perfectly, but I will try to change the settings and see if it helps... Thanks!

StuartB4, 2020-10-22:
Ok.

closed 2020-12-16 by arkypita