arkypita/LaserGRBL #1146: firmware-problem for my ORTUR LaserMaster 2 20w

GuidoMonstrey, 2020-10-23:
Hi, Arkypita, can i ask you a question please?
I have a problem with my machine. Will you read below ?
[image]
[image]

Machine : ORTUR LASER MASTER 2 20W (bought it on 30 october 2020) (it says at the lightburn and LaserGRBL that the firmware is OLV_137.BIN, i think)

software: Lightburn or-and LaserGRBL

PC :HP envy 17ca1004nb with  Windows 10

 

The machine did work almost oke. Just when i was engraving, i did get some lines that did not match together…there was always a difference between the left and right side of a ’round sign’ (as example). The left stood higher then the right…also with text, the letters were not at the same line…so, i was thinking it is maybe the firmware (OLV_137.BIN i think). Altough i did read at the ORTUR-website not to change the firmware when the machine is older then mars 2020, i did….i was searching where it came from. It was with Lightburn and LaserGRBL the same. And now i have big problems.  🙁

So my question is :

Can i re-install the factory-settings (with i think OLV_137.BIN) ?

I  have made  a mistake to install and i did install the firmware OLV_134.BIN.

And now my machine is not working correct anymore: the laser does not burn anymore (no blue light). When i start the machine(power on) he goes to another startpoint and makes some noises with his servo’s, because he can’t go any further, then it stops by himself.

Can i find maybe the correct firmware and software for this machine please? (OLV_137.BIN) or is my machine broken?

Thank you for taking your time for reading this email.

Guido Monstrey Varsenare Belgium

arkypita, 2020-10-23:
Good morning

I don't think your problems have anything to do with firmware. Firmware is not something that the user can modify, that can be ruined over time, that can stop working. It is something that is electronically programmed into the control board chips and that, unless you specifically want to do, it never changes.

I think your problems are instead mechanical or configuration problems.

Imperfections in alignments are definitely due to something like loose screws, loose belts, or too high speeds that cause the motors to lose steps.

First of all check your hardware.

If you didn't solve you can configure firmware (you can do that with LaserGRBL from menu "grbl configuration").
You can set lower speed to acceleration and max speed. Read: http://lasergrbl.com/configuration/

**Before changing anything save the current configuration (export)**

arkypita, 2020-10-23:
For specific ortur support contact its official facebook comunity or helpdesk

https://www.facebook.com/groups/orturusergroup
https://ortur.tech/support/

GuidoMonstrey, 2020-10-24:
Hi arkyita, thank you for your message. It was as you said a mechanical problem. The belts (both) for the y-axis were not tight enough.
I did tighten them up and the machine was working perfect. So the belts must be very tight.
Also, i did get the firmware (OVL137.BIN) from the super-manager by Ortur on facebook.
So, my problems are solved thanks to you both.....
Thank you again for your superhelp.
Grts
Guido Belgium

closed 2020-12-16 by arkypita