arkypita/LaserGRBL #1147: WiFi connection

ssombra, 2020-10-24:
I am trying to connect Lasergrbl via wifi so I got one of those Uno + wifi boards:

[image]

I followed the instructions in the [Lasergrbl webpage for wifi connection](http://lasergrbl.com/usage/wifi-with-esp8266/): I installed the libraries and the boards definitions, I can upload GRBL to the Uno side of the board, and it seems that I can also upload the sketch to the ESP8266 side of the board:

[image]

After that, I reboot the board and I can see the ESP8266 wifi but everytime I try to connect to it I get an error message that I can't connect to that network. I tried with the telnet sketch, with the Websocket sketch and I even tried with the Laserweb sketch and always the same problem. I think that it is a problem with the libraries. Are the instructions still valid for the current board definitions and libraries? should I try with older versions? anyone with this board was succesful?

gmmanonymus111, 2020-10-24:
I had exactly that board for a while. I found a telnet bridge library (I can't recall it where), that was working okayish (I had to lower status updates, and some processed gcodes stuck in LaserGRBL's buffer, the connection sometimes just dropped because esp buffer got full). After this, I found [ESP-Link](https://github.com/jeelabs/esp-link) which is similar to the telnet bridge, but easier to config, more stable, and supports multiple connections at the same time. Flashing it is a bit harder, because you have to use esptool command manually, but that is documented in their wiki.

ssombra, 2020-10-24:
> esptool

> I had exactly that board for a while. I found a telnet bridge library (I can't recall it where), that was working okayish (I had to lower status updates, and some processed gcodes stuck in [...]

Thanks for the reply. Will try to take a look at the documentation, but this seems way beyond my understanding.

gmmanonymus111, 2020-10-24:
If you need a step-by-step tutorial, how to flash it, I'll try to write an easy one, it might be useful for other users

ssombra, 2020-10-24:
> If you need a step-by-step tutorial, how to flash it, I'll try to write an easy one, it might be useful for other users

That would be great, thanks for the offer.

arkypita, 2020-10-24:
I have exactly the same board. I use it without any problem with the telnet sketch, since 2017.
I haven't needed to recompile it since I installed it, so I don't know if the libraries it uses are still compatible.

However, the sketch itself does not nothing else that read from wifi and write to serial, and vice versa.
All the WiFi stuff is managed by the libraries.

ssombra, 2020-10-25:
> I have exactly the same board. I use it without any problem with the telnet sketch, since 2017.
>
> I haven't needed to recompile it since I installed it, so I don't know if the libraries it uses are [...]

I am sure my problem is not with the sketch but the libraries as I tried 3 different sketches. Maybe it is the way I upload the sketch to the ESP8266. Maybe you don't remember now, but I do the following after adding the libraries to the ARDUINO ide: 
1) put switches 5, 6 and 7 in the "on" position and upload the sketch to the ESP8266 using the ARDUINO IDE
2) Then I reboot the board with all switch positions I can imagine 
3) sometimes The ESP wifi is not visible and the times they are, I get the can't connect message when trying to connect to it 

Is there something I am missing? Should I need to do something else to the ESP?

ssombra, 2020-10-27:
> I had exactly that board for a while. I found a telnet bridge library (I can't recall it where), that was working okayish (I had to lower status updates, and some processed gcodes stuck in [...]

I was able to flash ESP-Link following the instructions in the site and using ESPTOOL, but I still have the same problem, no Access Point at all or, in the best case, it appears but it is not possible to connect to it. I have actually 2 of these boards and I have the problem with both. So still no luck for adding wifi to my Laser engraver with these boards.

gmmanonymus111, 2020-10-27:
It's sad to hear you are still facing this issue. Could you try erasing all data on your board? (`esptool.py --port {YOURUSBPORT} erase_flash`) There might be a corrupted setting there, and by default, this region is ignored when flashing new firmware.

arkypita, 2020-10-28:
Yesterday I spent the whole evening trying to program an esp-01 with the sketch, but it seems that I forgot how to do it because I was unable to upload the sketch to the chip!

I'll try again in the next few days

ssombra, 2020-10-28:
> It's sad to hear you are still facing this issue. Could you try erasing all data on your board? (`esptool.py --port {YOURUSBPORT} erase_flash`) There might be a corrupted setting there, and by default, this region is ignored when flashing new firmware.

I tried this but the problem persists. I think I will give up with this board and get some cheap ESP-01s modules. It is a shame because I really liked the idea of having all integrated in one board.

ssombra, 2020-10-28:
> Yesterday I spent the whole evening trying to program an esp-01 with the sketch, but it seems that I forgot how to do it because I was unable to upload the sketch to the chip!
> 
> I'll try again in the next few days

I can't even tell you how many hours I spent on the board :). I can follow tutorials and guides but my electronics knowledge is very limited, so it takes me forever to troubleshoot something.

arkypita, 2020-10-28:
I was able to flash esp-01 (tested with blink sketch) then with telnet sketch.

Unlucky cannot find any wifi connection created by esp. Try with new and older version of library (both esp board library and wifimanager downgrader and upgraded). For me is a mistery.

More test next days.

Please keep in mind that esp is current hungry device, do not power up the board with usb only. Plug power supply.
Board like yours need proper different dip-switch configuration for flash and for run. Check board documentation.

Based on lgrbl telemetry 86 user use LaserWebESP8266, 50 use Telnet. Not so much but prove it can work.
Keep trying

ssombra, 2020-10-29:
> I was able to flash esp-01 (tested with blink sketch) then with telnet sketch.
>
> Unlucky cannot find any wifi connection created by esp. Try with new and older version of library (both esp board [...]

The same happens to me, I can flash the board but then I can not find the Access point created by the ESP or, in some very rare cases I see the Access Point for a while (then it disappears) but I can't connect to it. I actually tried with an external power supply (12v 2A) and it is the same. I just tried flashing the telnet sketch to a ESP-01s module and it works as expected, so the problem should be with the UNO + ESP8266 board itself. I still would love to use that board, so I am looking forward for any development you could get. 

By the way, I connected the ESP-01s module directly to an UNO board without resistors or voltage dividers (taking advantage of the 3.3v pin in the UNO): RX to TX, TX to RX, Reset to RST, 3.3v to 3.3v and GND to GND and it works well. Why are the resistors connected to RX-TX in your schematics for?

arkypita, 2020-10-29:
> What are the resistors connected to RX-TX in your schematics for?

Arduino is a 5v device, esp8266 is limited to 3.3v.
Resistor act as voltage divider to protect esp rx pin.

ssombra, 2020-10-29:
> > What are the resistors connected to RX-TX in your schematics for?
> 
> Arduino is a 5v device, esp8266 is limited to 3.3v.
> Resistor act as voltage divider to protect esp rx pin.

I see. thanks.

gmmanonymus111, 2020-10-29:
## How to program Uno r3+wifi microcontroller for LaserGRBL

#### 1. ESP8266 programming with WiFiAccessPoint example sketch

- First, make sure your switches are set to the following:

| Switch | 1  |  2 | 3  | 4  | 5  | 6  | 7  | 8  |
| --- | --- | --- | --- | --- | --- | --- | --- | --- |
| **State** | OFF  | OFF  | OFF  | OFF  | ON  | ON  | ON  | -  |

This way it is set to programming mode, and the CH340 is connected to the ESP chip.

------------

I'm going to assume you have added ESP8266 boards to Arduino IDE (In case I'm wrong, there is a tutorial for that, near the end of this document).

- Once you have launched Arduino IDE, set your board type as `Generic ESP8266 Module` (`Tools` -> `Board` -> `ESP8266 boards` -> `Generic ESP8266 Module` in Arduino IDE 1.8.13).
[image]
- Set CPU Frequency to 80Mhz
- Set Crystal Frequency to 26Mhz (40Mhz caused errors for me, the board kept restarting)
- Set Upload Speed to 115200
- Set the correct COM port
- The rest can be set as seen on the image above

- Click `Upload` button. This should upload the sketch to the board
[image]
- Once done, set the switches as the following (#7 switch is set to OFF):

| Switch | 1  |  2 | 3  | 4  | 5  | 6  | 7  | 8  |
| --- | --- | --- | --- | --- | --- | --- | --- | --- |
| **State** | OFF  | OFF  | OFF  | OFF  | ON  | ON  | OFF  | -  |

This way CH340 is connected to the ESP chip, and you can see Serial Monitor output in Arduino IDE

- **Restart your board!**
- After a few seconds, a WiFi point should show up with the name `ESPap`
[image]
- You can connect to it, the default password is `thereisnospoon`
- Once connected, you can check your connection by typing `192.168.1.4` in your favourite browser's address bar. You should see a similar webpage:
[image]
- If eveything is working, you can continue to flash [ESP-LINK](https://github.com/jeelabs/esp-link "ESP-LINK")

------------

------------

#### 2. ESP8266 programming with [ESP-LINK](https://github.com/jeelabs/esp-link "ESP-LINK") and [ESPTool](https://github.com/espressif/esptool "ESPTool")

- First, make sure your switches are set to the following:

| Switch | 1  |  2 | 3  | 4  | 5  | 6  | 7  | 8  |
| --- | --- | --- | --- | --- | --- | --- | --- | --- |
| **State** | OFF  | OFF  | OFF  | OFF  | ON  | ON  | ON  | -  |

This way it is set to programming mode, and the CH340 is connected to the ESP chip.

------------

Again, I'm going to assume you have downloaded ESPTool (In case I'm wrong, there is a tutorial for that, near the end of this document).

- First, download [the latest ESP-LINK release](https://github.com/jeelabs/esp-link/releases/latest "ESP-LINK")
- Once downloaded, extarct it to somewhere. I'm goin to use esp8266 folder on my desktop
- Open Command Prompt in the folder you extracted the files into.
(TL:DR one-liner to flash eveything: `esptool.py --port COM4 write_flash 0x00000 boot_v1.6.bin 0x3FE000 blank.bin 0x3FC000 esp_init_data_default.bin 0x01000 user1.bin`)
- First, you have to flash a bootloader. I'll use the one included the archive. The command should be `esptool.py --port COM4 write_flash 0x00000 boot_v1.6.bin`
**Make sure to set the correct COM port!** If you get an error as `PermissionDenied` while uploading, make sure you have closed the Serial Monitor in Arduino IDE
[image]
After it was closed:
[image]
- **Reset your board!**
[... 53 lines cut ...]
[MSG:'$H'|'$X' to unlock]
```

-------

-------

#### 4. Connecting ATmega328P with ESP8266
- You only need to set the switches as the following:

| Switch | 1  |  2 | 3  | 4  | 5  | 6  | 7  | 8  |
| --- | --- | --- | --- | --- | --- | --- | --- | --- |
| **State** | ON  | ON  | OFF  | OFF  | OFF  | OFF  | OFF  | -  |

- When done, ESP-link should show you GRBL command responses while you are accessing it through your browser, or through telnet, or through LaserGRBL
[image]

-------

-------

#### 5. Bonus: installing ESP boards and ESPTool

- For ESP8266 boards, see [ESP8266 installing with boards manager](https://github.com/esp8266/Arduino#installing-with-boards-manager "ESP_Arduino")
- For ESPTool, it is just a simple Python command: `pip install esptool`. You can download Python from [here](https://www.python.org/downloads/ "here")

ssombra, 2020-10-29:
Thanks for taking the time for this very comprehensive tutorial. I followed it and I get stuck in the same place, even with the WiFiAccessPoint example:

[image]

The sketch uploads, the Access Point shows in the network list, but then I can´t connect to it. I didn´t go to the ESP-link / Esptool part because I keep getting the same no connection error. I bought two of these boards and both have the same problem, I know that it is very unlikely, but maybe both are defective, who knows.

gmmanonymus111, 2020-10-29:
Can you connect to the ESPap using any other device (e.g. a smartphone)
A user mentioned Intel driver issues at [ESPDeauther](https://github.com/SpacehuhnTech/esp8266_deauther/issues/754), but they were able to connect to the AP with a smartphone
From here on, I'm only guessing :/

ssombra, 2020-10-29:
Actually I had edited my previous comment to say that the access point doesn't even appear in my iPhone, not a single time from all the tests I made.

gmmanonymus111, 2020-10-29:
Oh, didn't see that. Found [another issue](https://github.com/esp8266/Arduino/issues/1094), where they explain that esp8266 has only one radio, and it gets bugged a lot because of cached settings, it can helped by calling `ESP.eraseConfig(); `

ssombra, 2020-10-29:
Thanks for the links, I checked them and tried all the ideas there (eraseConfig, disconnect, changing phy mode) and none worked. Will keep looking for info.

harryberlin, 2020-11-15:
i think wifi connection has lot of risk.
is there a watchdog to stop and hold the system, if connection is lost?

gmmanonymus111, 2020-11-15:
Not yet, but I think WiFi connection is very stable (for me, there were no accidental disconnects at all), and for a small machine like this, the biggest problem could be you burn a big black dot, if the laser stays on, and you didn't hear the disconnect sound. For bigger, more serious machines, it is better to use wired connection of course.

arkypita, 2020-11-15:
@harryberlin the same risk you have with USB.
For what I see only Ortur has implemented a sort of watchdog in their firmware that shut down the laser if USB is disconnected. Standard grbl machine doesn't do anything.

Moreover WiFi allow you put the engraver outside, far from PC where nothing can burn, instead inside you home on the same table you put your PC.

closed 2020-12-16 by arkypita