arkypita/LaserGRBL #1149: Reversing second Y-axis direction

nbonham2, 2020-10-25:
Hello all,

I bought an engraver from Two Trees the uses to Y-axis motors. I want to reverse the direction of one of the motors so they go in tandem with each other rather than twisting the X-axis brace. I saw in the configurations where to reverse directions but none of the codes had a 2 Y-axis option. Can anyone help?

StuartB4, 2020-10-25:
Could you not just rotate the plug of one of the motor 180 degrees. 
If the motors shafts face each other, one needs to go clockwise and the other counter clockwise.
Rotating the plug should do that.

arkypita, 2020-10-25:
LaserGRBL has no control over the direction of the motors, it simply sends g-code commands that are executed by the board.
It's the board itself (grbl firmware running on Atmega328 chip) that translates the commands into impulses for the motors.

So this is not the right place to ask this question.

To my knowledge, not even grbl have that kind of control, because there is no "double motor" concept in grbl firmware.
When using two motors per axis, usually the motors are controlled by the same output pins.

What you want to do must be solved electronically. The chip that drive the motors (A4988, DRV8825 or similar) it has a pin that controls the direction (usually called DIR).

[image]

You have to cut the path that reaches this pin and insert an inverting logic, for example with a transistor.

nbonham2, 2020-10-26:
> Could you not just rotate the plug of one of the motor 180 degrees.
> If the motors shafts face each other, one needs to go clockwise and the other counter clockwise.
> Rotating the plug should do that.

The plugs and motors have guides that only allow them to be plugged in 1 direction. I was thinking about just simply flipping the entire motor/carriage assembly 180 degrees so that it faces the same direction as it’s opposing motor. I would just have to cut the rail down an inch or two and lose that much work area

nbonham2, 2020-10-26:
> LaserGRBL has no control over the direction of the motors, it simply sends g-code commands that are executed by the board.
> It's the board itself (grbl firmware running on Atmega328 chip) that [...]

What your saying makes perfect sense in my head. But I don’t trust myself enough to do that kind of work on the board, nor do I have the proper tools to do it. I might try to find some videos about doing what you’re saying and seeing if it would be worth my time trying to get it to work. The engraver kit wasn’t all that expensive, so I wouldn’t be risking a whole lot if I messed up

arkypita, 2020-10-26:
I have a DIY engraver with double Y motor. Like you, my motors has to rotate in opposite directions.

[image]

I use Protoneer CNC v3 shield. I don't remember to have done something special to obtain this result, but I have built it long time ago. As my board does not have plugs, and I had to solder my own wiring, I guess I simply invert motor wiring.

If you cannot rotate the plug, just cut wires and swap them.
Or build an adapter.
Or better, remove the pins from the plug and reinsert them swapped.

nbonham2, 2020-10-26:
> Or better, remove the pins from the plug and reinsert them swapped.

I would prefer to do that. Is it as easy as just flipping the pins around? Sorry for being naive, I’m relatively new to this kind of stuff. Thanks for your help.

StuartB4, 2020-10-26:
I just sliced the tabs off with a sharp blade so the plug fits either way now. 

Or do this. 

[image]

arkypita, 2020-10-26:
Pins have a metal flap that holds them in place. Press it gently with an awl or blade tip while pulling the wire.
See some tutorial, practice on some scrap if you have any.

https://www.youtube.com/watch?v=S9xeg_3eREc

arkypita, 2020-10-26:
According to this [1] to change direction you need to reverse the current in one set of coils only.

[1] https://electronics.stackexchange.com/questions/70643/how-to-reverse-rotation-direction-of-stepper-motor#:~:text=To%20change%20direction%20you%20need,with%20the%20same%20drive%20signal.

BeatSlayer, 2020-10-26:
Reversing the connector had the same effect as reversing the coils 3 times, as 2 times cancel each other, you end up with one coil reversed.

closed 2020-12-16 by arkypita