arkypita/LaserGRBL #1153: New ORTUR Laser Master 2 with Alarm

Anto217, 2020-11-01:
Dear All, 

Hope you are doing great. 

Wanted some help with you. I am a beginner. Bought my Laser Master  2 (Ortur), got it, assembled it, dowloaded Laser GRBLv3.8.0 and the custom buttons. 

The software communicates with the hardware, but when it connects, the right hand corner status says "Alarm" and the left hand corner says {MSG: Check Limits}

Also, One of the parts came misassembled. Please guide me!! Do you know what the problem is? and how to fix it?

[image]
[image]

Anto217, 2020-11-01:
[image]

arkypita, 2020-11-02:
LaserGRBL is a free software downloadable from the internet, not tied to a specific product.
I suggest you to contact the specialized assistance of your engraver.

For ORTUR brand:
- [Ortur user group (facebook community)](https://www.facebook.com/groups/orturusergroup)
- [Ortur customer support](https://ortur.tech/support/)

Best regards
Arkypita, developer of LaserGRBL

closed 2020-12-16 by arkypita