arkypita/LaserGRBL #1156: LaserGRBL PWM

mike8760, 2020-11-03:
I recently bought a small laser engraver that came with a no-name controller, which I think is an EleksCAM Mana or clone. The connection for the laser is two wires, and I assumed it would be ON/OFF control, so I un-checked the PWM function in GRBL; this made no difference. 'Scoping the power to the laser I see that is some sort of PWM, and the laser never runs at full power, which is needed since it's only 500mW!
I'm puzzled, both by the GRBL laser control and the controller connection; there is an un-populated socket for "laser" which has 3 pins and which I have assumed would be for a PWM connection.
I'd be obliged if someone more familiar with LaserGRBL, and such clone controllers, could shed some light on my problem. I should add that I'm a complete newbie to the world of laser engravers...

gmmanonymus111, 2020-11-03:
Could you upload some pictures of your driver, laser and current connection?

mike8760, 2020-11-03:
Thanks for your response.  Below are (links?) to the controller and laser.
[image]
[image]
Not sure what you mean by the current connection.  So far I've found that the laser mosfet driver is connected to the Nano pin D11, which seems to be (on some diagrams) variable spindle PWM. The laser is controlled to some extent 'correctly' as it is turned ON and OFF when appropriate.  The rest of the setup is fairly common stepper motors; I intend to fit limit switches at some time as I am sconfused by the various sizing and scales used for plot areas.
Regards.

gmmanonymus111, 2020-11-03:
Thank you for the images!
By current connection, I meant the wiring of your setup (kind of incorrect words, sorry for that).
It seems your laser is connected to a simple 12V socket (at least according to EleksMana's board)
You should have a 3-pin socket and connector, to be able to use PWM.
You are correct as this seems it can only operate in ON/OFF mode.
[image]
 It should output 12V, if you find the laser beam weak, measure this voltage. Also, make sure you use at least a 2A PSU. In case all above is correct, maybe the built-in driver won't let you use 100% of the output power.
Did the laser come with this set-up bundled? Does it have a 3rd wire for pwm control?

mike8760, 2020-11-03:
Thanks for taking the time for your assistance. The illustration you show is similar to mine, but clearly not a Mana SE. The problem seems be that the only socket for the laser is the wrong one, and it should be connected to the 3 pin socket shown in your diagram, although my laser does not support PWM  So, a possibility is that the 'makers' intended to supply a Nano with a modified grbl code to support a simple ON/OFF signal for the laser on pin D11 - but actually supplied the wrong one. I imagine it is fairly simple to do this modification, as the source code is available.  Perhaps I could consider this, but I don't have an AVR C compiler!!
Any ideas?

gmmanonymus111, 2020-11-03:
As far as I know, GRBL for Arduino Nano can be compiled and flashed with Arduino IDE.

mike8760, 2020-11-04:
Of course, thanks for reminding me!  I have not used Arduino stuff very much at all; I guess I'm in the Microchip PIC camp.
For me, the question is, what might be changed in grbl to allow my laser connection to be ON/OFF, but with control maintained for when it should be OFF for fast moves. I see that on my controller the 3 pin socket is intended for TTL PWM, as the control line is also connected to Nano pin D11.  It isn't a rabbit-hole I had anticipated when I got the laser engraver, as I am only interested in making the occasional PCB by paint ablation and etching.  Perhaps I should look for another program, like Flatcam, and see how that behaves.
Thanks for your time.

mike8760, 2020-11-17:
It seems I am still having problems with my setup. One thing I had not considered before, but that is the minimum PC system requirements for LaserGRBL?   I have trawled many websites but so far have found nothing with repect minimum RAM, Disk space, OS (I now know XP is verboten!) or USB level etc.
Perhaps someone with more experience could enlighten me?

StuartB4, 2020-11-17:
LaserGRBL does work with XP, as long as the Microsoft's .Net Framework is the correct version.
It needs version 3.5 or higher. 

You don't need a big fancy PC because LaserGRBL is a streaming program. It just need's to be able to stream the gcode from the PC to the laser engraver control board.

arkypita, 2020-11-17:
[image]

mike8760, 2020-11-18:
Thanks for taking the time to answer to what must seem silly questions.  One reason I was looking for minimum system specs was concerned with available RAM.  The old Dell laptop I wished to dedicate to laser stuff has 2Gb RAM, and LaserGRBL issues the error message "program too big to fit in memory". (I only manage to see this if I attempt installation via Powershell).  I can only increase the memory in this laptop to 4Gb, but I am doubtful if this would still be inadequate?  I see the size of install.exe is over 4Gb, but the lasergrbl.exe is less than 3Gb.  
Any ideas?

gmmanonymus111, 2020-11-18:
I have an Asus Eee PC 1011PX (a really old netbook, with only 2GB RAM) with Win7 installed on it, and LaserGRBL runs just fine. How did you manage to get that size of the installer? Did you mean RAM consumption?

arkypita, 2020-11-18:
> I see the size of install.exe is over 4Gb, but the lasergrbl.exe is less than 3Gb.

What? Installer is 4MB, LaserGRBL use 26 MB ram when running.

[image]

[image]

arkypita, 2020-11-18:
LaserGRBL only needs memory when dealing with large images, so it can safely use 2-3GB of memory.

If your computer has less than 2GB of memory this is not a problem, because the operating system makes virtual memory available by swapping memory to disk (of course this mean lower processing performance).

closed 2020-12-16 by arkypita