arkypita/LaserGRBL #1158: CNC 3018 Pro Laser Problems

locovaldes, 2020-11-03:
Hi i have a 3018 Pro , and the laser was working good and now sometime burn and sometime nothing, or do like small draw no normal
[image]

locovaldes, 2020-11-03:
first was working fine 
[image]
[image]

and now i try to do the 800 speed test and see what i got

locovaldes, 2020-11-03:
first was and now i try to do the 800 speed test and see what i got

locovaldes, 2020-11-03:
and sometime no burn nothing

gmmanonymus111, 2020-11-03:
While importing your image/vector, did you set the correct `S-Min` and `S-Max` values?
While importing images, did you set the correct raster settings?

(NOTE: the values on the pictures are for reference, random values)
[image]
[image]

locovaldes, 2020-11-03:
Hi if i use LaserGRBL where can i see that

locovaldes, 2020-11-03:
i importing a G code

locovaldes, 2020-11-03:
I did it this yesterday
[image]

locovaldes, 2020-11-03:
Right now not good
[image]

gmmanonymus111, 2020-11-04:
How old is your machine? It might be your diode is dying.
It's also worth to check the wiring connection, there might be some electrical noise, preventing your laser driver to receive correct PWM signal. Check your power supply, and make sure it is at least 2A.

locovaldes, 2020-11-04:
Is a new machineSent from my T-Mobile 4G LTE Device

BeatSlayer, 2020-11-05:
Are you focusing it?

locovaldes, 2020-11-05:
Yes thanks I already resolve the problem.Sent from my T-Mobile 4G LTE Device

closed 2020-12-16 by arkypita