arkypita/LaserGRBL #1160: Connection problem

hixnhx, 2020-11-06:
bought an old computer with new Windows 10 install and i cannot get it to connect. Everything is working with old machine. Any thoughts?

StuartB4, 2020-11-06:
Have you tried installing the driver on the Tools menu in LaserGRBL.

hixnhx, 2020-11-06:
Yes any other ideas

gmmanonymus111, 2020-11-07:
Any error messages?

hixnhx, 2020-11-07:
no error message. When I try to connect it says connect then after a while it goes to disconnect I went into the bios and it seems to be OK

Techman750, 2020-12-05:
I am having exactly the same problem, have tried different computers, re installed software, reinstalled driver tried different cables, tried different usb sockets run out of options now

arkypita, 2020-12-05:
> I am having exactly the same problem.

@Techman750 other user was able to use the marker on an old PC, and is having trouble on the new one. From what you describe instead you have never been able to make it work. Is it?

arkypita, 2020-12-05:
What kind of engraver do you have? Does it come with any form of instructions? Does it come with any proprietary software included?

Techman750, 2020-12-05:
It was a £120 3000mW Mini gravure Laser from Amazon (Chinese import I would think) no real instructions. It did work at first but after I updated the software it will not connect

arkypita, 2020-12-05:
> It did work at first but after I updated the software it will not connect.

If it worked with an older version of LaserGRBL, and now after an update of LaserGRBL it doesn't work anymore, you can just reinstall the [old version](https://github.com/arkypita/LaserGRBL/releases). Anyway I'm 100% sure it's not a LaserGRBL problem.

Keep this in mind:
- LaserGRBL is open software, freely downloadable from internet for free.
- LaserGRBL is not related to any specific laser engraver hardware or vendors.
- LaserGRBL come with no warranty and no support: neither as regards the software, nor even more so as regards the hardware.

I cannot guarantee that it will work with all engraver, from all manufacturers, all over the world. If you have received a copy of LaserGRBL in bundle with your purchased engraver, please consider asking support to vendors or manufacturer.

> It was a £120 3000mW Mini gravure Laser from Amazon

https://www.amazon.it/engraving-machine-Engraver-engraver-7-5X7-5CM/dp/B07H2WKZJ2

If it is this one, I see so many negative reviews.
From the amazon page I can't even figure out what software it should use. 

As far as we know, it may simply be broken.

We can try to help you figure out what's wrong, but you need to give some more information.
Have you tried using any other software?

hixnhx, 2020-12-05:
I believe my problem is the computer I purchased is to old to support windows 10 and the manufacturer does not produce the drivers that will connect. I gave up an took my losses used my newer laptop instead.

closed 2020-12-16 by arkypita