arkypita/LaserGRBL #1161: LaserGRBL

MattTonning, 2020-11-07:
I installed LaserGRBL the program opens but will not go any further.  I right-click import CustomButtons then the error comes back to say it cannot find CustomButton.GZ. What can I do next to make this software work?
[image]

gmmanonymus111, 2020-11-07:
To import custom buttons, you first need to download a button package, then navigate to the download location (or navigate to where you exported your previous buttons). More [here](http://lasergrbl.com/usage/custom-buttons/)

To use LaserGRBL, you need to have a GRBL controlled laser engraver (I'm going to assume you have one). To connect to you engraver with a USB cable, first connect your machine to your PC (and also make sure you have installed the proper drivers for your engraver, CH340 one can be found in LaserGRBL: `Tools` -> `Install CH340 driver`)
[image]

- To connect, you need to find the correct COM port (if you have only one, most likely it is your machine), and select it at LaserGRBL's start screen. Then, set your BAUD rate, it is most likely `115200`, then click connect.
[image]

- To connect wirelessly (for this, you of course need a wifi capabilities), go to `Grbl` -> `Settings` -> `Protocol` and set connection protocol to either `Telnet` or `LaserWeb` depending on your connection type.
[image]
Then, type the IP address of your machine (in case of `Telnet`, also use port):
[image]

Once done, click connect. (the plug icon near the IP address or the BAUD rate, depending which method you used)
A GRLB welcome message should appear:
[image]

Also, it is advised to read [LaserGRBL's FAQ](http://lasergrbl.com/faq/)

MattTonning, 2020-11-09:
Hi Again,

I am fully frustrated with this tool and the software. I have followed the
instructions you sent me but to no avail.
The software will not recognize the laser burner. Or even allow any manual
input. This is as far as I can get (see Pic attached)
Is it possible the software is not compatible with my windows 7 64? Since
it will not behave to your instructions?
I would love to get this thing working If not
[image: stumpted.jpg]
 I will send it back and look for a better product.

closed 2020-12-16 by arkypita