arkypita/LaserGRBL #1162: laser 30 Watt

wiwan1971, 2020-11-08:
Ciao a tutti, ho un laser da 30 watt nel database materiali non ce un laser 30 watt ma solo fino a 20 come posso aggiungerlo per dargli più potenza, grazie

arkypita, 2020-11-08:
Ciao
La potenza del laser è controllata esclusivamente dal valore S-Max nella finestra di impostazioni di import immagine.

[image]

Il database materiali non è altro che una scorciatoia per richiamare valori pre-impostati di potenza e velocità secondo dati forniti dal produttore. Al momento Ortur è l'unico produttore che ha fornito dei dati per i suoi prodotti.
Se trovi delle combinazioni di velocità e potenza che vanno bene per la tua marcatrice, sui vari materiali, puoi aggiungerli tu stesso al database, basta aggiungere delle righe all'elenco.

wiwan1971, 2020-11-08:
Grazie x l'informazione ciao

closed 2020-12-16 by arkypita