arkypita/LaserGRBL #1163: LaserGRBL cutting area preview

kukutsi, 2020-11-08:
Hi, I'm new to this, so excuse me if I'm asking something trivial.
I'm using an atomstack 20w laser engraver and LaserGRBL to engrave things.
I'm coming from a smaller laser engraver with its own engraving software (Laser Engraving Machine K4. V2.2).
I find difficult to find the exact position I have to put an item (for example a piece of wood) to engrave it symmetricaly.
How can I make the laser beam/pointer preview the engraving area, so as to align the item and put it exactly on the right position?
Thank you in advance for your time!!!

arkypita, 2020-11-08:
http://lasergrbl.com/usage/custom-buttons/

kukutsi, 2020-11-08:
O.K., thank you very much!!!

closed 2020-12-16 by arkypita