arkypita/LaserGRBL #1165: TTL connection for Laser GRBL

Hari-Merla, 2020-11-12:
hi,
i am trying to make a Laser Engraver.
i have got most things figured out.
the very last bit of the build is left.

I have been trying to turn on an off my Laser using the TTL Driver board, i have connected it to the Z+ pins on the cnc shield but my laser doesnt seem to be turning on.

Please Help me out with this.

Thank You

gmmanonymus111, 2020-11-12:
Could you upload some pictures about your setup and wiring?
What version of GRBL do you use?

Hari-Merla, 2020-11-12:
> Could you upload some pictures about your setup and wiring?
i will attach the photos of the wiring from CNC Shield to Laser Driver ,  i have disassembled the rest because i want to figure out this TTL connection. 

[image]
[image]
[image]
[image]

> What version of GRBL do you use?
i use the 1.1h version , i flashed the firmware using the LaserGRBL v3.8.0

StuartB4, 2020-11-12:
Are you sure the blue wire is ttl - and the green ttl + on the laser driver board. 
Do you have Support PWM enabled in Settings on the Grbl menu. Second Tab from the left. 
Also $32 set to 1 and $30 set to 255 in Grbl configuration on the Grbl menu.

StuartB4, 2020-11-12:
Does the 12 +/ - on the laser driver board next to the black power supply wires go to the laser fan, and the one next to the ttl socket to the laser diode?

Hari-Merla, 2020-11-12:
> Are you sure the blue wire is ttl - and the green ttl + on the laser driver board.
if i flip the green and the blue wires the laser stays for the whole time.

> Do you have Support PWM enabled in Settings on the Grbl menu. Second Tab from the left.
yes i have enabled the PWM in the setting.

> Also $32 set to 1 and $30 set to 255 in Grbl configuration on the Grbl menu.
Done that.

> Does the 12 +/ - on the laser driver board next to the black power supply wires go to the laser fan, and the one next to the ttl socket to the laser diode?

YES it does

StuartB4, 2020-11-12:
Ok.

gmmanonymus111, 2020-11-12:
Are the laser driver and the shield on common ground or do you use the same PSU for arduino/shiled and the laser driver?

Hari-Merla, 2020-11-12:
i have connected a volt meter to the Z+ pins and see that when the laser is off it is around 3v and when it is engraving it goes upto 7v,
Did the above without the Laser connected, just wanted to check whether the TTL Signal works

i wanted to ask do we need to use a resistor of any kind for the ttl connection.

Hari-Merla, 2020-11-12:
> Are the laser driver and the shield on common ground or do you use the same PSU for arduino/shiled and the laser driver?

There is no Ground option on my Shield and havent given Arduino any Ground.

PS: I am really thankful for your help and support.

gmmanonymus111, 2020-11-12:
> There is no Ground option on my Shield and havent given Arduino any Ground.

I meant the `-` pin as ground (and `+` is either 5V or 12V)

arkypita, 2020-11-12:
> i have connected a volt meter to the Z+ pins and see that when the laser is off it is around 3v and when it is engraving it goes upto 7v,
> Did the above without the Laser connected, just wanted to [...]

Because Arduino is a 5v device you should read 0v with laser off and about  5v with laser at full power. Any different values mean something wrong in your circuit or in the mode you use the multimeter

Hari-Merla, 2020-11-12:
> I meant the `-` pin as ground (and `+` is either 5V or 12V)

they are on the same power supply but different connections

[image]

arkypita, 2020-11-12:
When you measure z+ the red multimeter cable should be connected to z+, the black one should be connected to any of the black stripline pins (they are all connected to ground) the multimeter shoul be settled to voltage measurements DC

Hari-Merla, 2020-11-13:
> When you measure z+ the red multimeter cable should be connected to z+, the black one should be connected to any of the black stripline pins (they are all connected to ground) the multimeter shoul be settled to voltage measurements DC

i have connected the Voltmeter to the Z+ and Z- pins 
the arduino is in idle state 
i am getting a reading of 2.31v

please find the Picture attached
[image]

arkypita, 2020-11-13:
Can you try with the CNC motor shield removed, only on the arduino UNO pins connected via its own USB?

Here is the pinout: https://github.com/gnea/grbl/wiki/Connecting-Grbl

arkypita, 2020-11-13:
Pin labeled GND for black probe, Pin labeled "11" (variable spindle pwm) for red probe.

Hari-Merla, 2020-11-13:
> Pin labeled GND for black probe, Pin labeled "11" (variable spindle pwm) for red probe.
yea i figured out it was the problem in the Arduino itself.
i am using a new one now and am getting the voltages correct,
hopefully my laser will work now.

Thank you sooo muchh for your time and help.

Hari-Merla, 2020-11-25:
Thank you so much everyone for the Time and help , i got my engraver to work.
if you are interested in knowing any details or the approach write to me on - harijagan9866@gmail.com.

Cheers!!!

closed 2020-11-25 by Hari-Merla

FiusRa, 2021-03-02:
Hello. my language is not english. it is a translation . may be bad. In general, I have a driver on the xl4015 microcircuit. in the datasheet, it can be controlled by ttl. here is the datasheet. https://datasheetspdf.com/pdf-file/786208/Xlsemi/XL4015/1. 8 page. the problem is that the management is reversed. that is, when the high level is turned off when the low is turned on. how to invert the output to the firmware. so that pin z is always high.

FiusRa, 2021-03-02:
Typical System Application (TTL shutdown function)
Logic level signals shutdown function can be used in typical system application with external
components. When the TTL high voltage above 3.3V(referenced to ground, lower than VIN), the
converter will shutdown, input current less than 5mA; when the TTL Low voltage below
0.8V(referenced to ground), the converter will turn on.