arkypita/LaserGRBL #1166: Other firmware version

gerdvogel, 2020-11-12:
I have a question regarding the Firmeware version of the Other Master Laser Engraver.
I have seen that GRBL 1.1f is installed and when I press §I to display the information I see the following.
$I
[AUTHOR: RenShen]
[BUILD: Orther Laser V1.2]
[DATE:2:23:00 - Feb 6 2020]
Ver:1,1f.20170801:]
[OPT:VZD,35,254]

Question: is this the Other Firmeware Version OLF_134 ?

I have a problem when laser cutting circles with G02 and G03. The lasered circles are not circular and my laser does not show a point but a rectangle.

Maybe one of them has an answer for both questions.

Thanks a lot in advance.

Greeting
Gerd

[image]

[image]

arkypita, 2020-11-12:
Non circular circles and square laser dot are not related to firmware version but with untighten belt and unfocused lens.

LaserGRBL is a free software downloadable from the internet, not tied to a specific product.
I suggest you to contact the specialized assistance of your engraver.

For ORTUR brand:

- [Ortur user group (facebook community)](https://www.facebook.com/groups/orturusergroup)
- [Ortur customer support](https://ortur.tech/support/)

Best regards
Arkypita, developer of LaserGRBL

gerdvogel, 2020-11-12:
Can you tell me which Other version I have on the laser?

See the image from LaserGrbl with the result of the $I command.

Best regards
Gerd

gerdvogel, 2020-11-13:
Thank you very much for the tip to re-tension the toothed belts. It helped and the circles are circular again.

Best regards
Gerd

closed 2020-12-16 by arkypita