arkypita/LaserGRBL #1167: LaserGrbl Wifi using Adafruit Feather HUZZAH with ESP8266 WiFi

Wallio007, 2020-11-14:
Hi,

I want to use this module : Adafruit Feather HUZZAH with ESP8266 WiFi to connect my GRBL board with wifi.
[https://www.adafruit.com/product/2821](url)
The board is a Gradus M1 Pro with GRBL : [http://www.panucatt.com/product_p/gm1pro.htm](url)

I uploaded websocket.ino program in the Adafruit Feather, did all steps to setup wifi.

In LaserGrbl, the line "connected" appears, but I'm not able to use any axis ... everything is stuck.

Do you have any idea what's happening here ?

Thanks for your help.

Thomas

arkypita, 2020-11-14:
screenshoot of lasergrbl connected?
what messages appare when you push jog buttons?

Wallio007, 2020-11-15:
Hi,

Here is a screenshoot : 

[image]

I cannot push any jog buttons

arkypita, 2020-11-15:
Hi @Wallio007 

"Connected" message in the message log (left box, under "type gcode here") is sent by [websocket.ino sketch from the ESP8266](https://github.com/arkypita/ESP8266-SerialWebSocket/blob/c3c5adc440c38a5fad937d72011d8ee7be9ca3bb/websocketserver/websocketserver.ino#L99) when a connection is detected, this is a good sign: half job is done.

however, there is one more thing that LaserGRBL expects to receive, and it is grbl's welcome message (Grbl v1.1f), together with the status messages it sends to it (via ESP8266) that put it in the "Idle" state.

The string "Connectè" is just a translator's error, it should have written "En connession" (connecting). In fact I think that after a few seconds your LaserGRBL closes the connection, because it does not receive signals from grbl.

Check your wiring between Gradus M1 and Adafruit Feather: TX goes to RX and RX goes to TX, be care to put some resistor to protect 3.3v RX line of ESP8266.

[image]

Wallio007, 2020-11-15:
Hi,

Thanks for you help, it seems to be working now.
I re-upload the sketch in Adafruit Feather, and connected Gradus M1 Pro and Adafruit Feather as follow :  

Gradus M1 Pro   <---->   Adafruit Feather
                 +5V   <---->  VBUS ( connected to +5V USB PORT)
                GND   <---->  GND
                  TX    <---->    RX
                  RX    <---->    TX

NB : I didn't use any resistor.

I will run some test later.

Thanks for your help.

arkypita, 2020-11-15:
Gradus board TX pin is a 5v output pin 
Esp8266 RX pin is 3.3v input pin.

N.B it can work without resistor voltage divider, but can burn esp pin in few days.

Wallio007, 2020-11-15:
Hi,

Yeah you are right, TX pin on Gradus board is +5V.
On Feather board : 
[image]

So it seems like TX from Gradus with +5V will be handle by Feather RX pin (see "RX pin 5V compliant" on the picture)

But how about TX on Feather with RX on Gradus ?

Thanks for your help 👍

arkypita, 2020-11-15:
So Feather has internal voltage divider to protect RX pin, no need of resistor.
TX pin is not a problem since 3.3v is recognized as high level from TTL logic.

[image]

Wallio007, 2020-11-15:
Great ! Thanks 👍

closed 2020-12-16 by arkypita