arkypita/LaserGRBL #1168: Continuous Jog only operating in one direction

WillBea, 2020-11-15:
The Continuous Jog, when selected, only operates in a positive direction. For example : If you press an arrow key to go right it moves right, if you press the left arrow it still goes right. Likewise if you press the arrow key to send the Y axis forward it goes forward, if you press the arrow key to bring the axis back it still goes forward. Also, a minor point, Continuous is spelt incorrectly, next to the selection box. It's spelt correctly in the description.

regpye, 2021-03-21:
I am having a similar problem, see general discussions **Strange homing problem #1377**
Narrowed the problem down to LaserGRBL because after many re-wires and testing I used UGS to do a test and everything works fine using that software.

arkypita, 2021-03-23:
I close here, I follow the problem in the other post. #1377

closed 2021-03-23 by arkypita