arkypita/LaserGRBL #1169: Image is burning and smudging

janemarie425, 2020-11-16:
Hello,

I am very new to LaserGRBL. I recently purchased an Ortur Laser Master 15W.  When I prepare an image and send to the laser is is burning black the image and smudging.  Only one time did send an image to the laser and it more like carved it and didnt burn it. It finished brown and appeared deeper and did not smudge.  Please help.  Thanks kindly.
[image]
[image]

StuartB4, 2020-11-17:
Start LaserGRBL and click connect. Check in Grbl configuration on the Grbl menu. and make sure Laser Mode is enabled. 
That is $32, set it to 1 if it's on 0. Check $30, it should be set to 255 ( or 1000).
If you have to change them click Write after and then Close. 

Then when you load an image and you get all the settings down the left side, adjust the Brightness and Contrast if you need to.
Then slide the White clip slider to the right until all the background interference around your image or text turns to pink. 

After you've done that and clicked Next make sure the S-Min setting is 0 and the and the S-Max is either 255 or 1000 to match whatever $30 is. Then you can adjust the power using S-Max for individual images. 

If Laser On in Target Image settings is set to M3 chnage it to M4.

[image]

[image]

BeatSlayer, 2020-11-17:
Less power and slower.

janemarie425, 2020-11-17:
Thank you so much!! I will give it a try. 

Kind regards,

Jane

closed 2020-12-16 by arkypita