arkypita/LaserGRBL #1171: Sugestion

Abysslabs, 2020-11-19:
First of all, THANKS for the great software.

After using this software, I noticed that the software has few options to perform image engrave and is a chalange to get good results, so in my opinion is mutch easier to edit the image in software like GIMP and export the image ready to use already in 1Bit(dither).
My sugestion is to have an option like "pass trough" to not touch the image, just use the original file has it is, since the image is already in 1Bit dither.
I think is a basic option to implement, I dont know.

arkypita, 2020-11-20:
Line 2 Line tracing with "Sharp" option for resize could do that for you.

[image]

Just be sure to compute size from exif in the target size option window, so the image is not scaled uselessly.

[image]

Not sure if this is really a "pass trough", but it is something that comes very close

BeatSlayer, 2020-11-20:
As long as you don't resize the image the method @arkypita told acts as a passthrough, it'll just read the pixel data line by line and calculate the moves and power.

Abysslabs, 2020-11-20:
Thanks for the info, i'll give it try.

MarkDowes, 2020-12-02:
Ah this is exactly what I was looking for yesterday, unfortunately I am having some crazy error using the "Sharp (nearest neighbour)" setting where chunks of my image are missing. Maybe I should try different file formats.