arkypita/LaserGRBL #1174: Cause : UnexpectedDisconnect

min0200, 2020-11-20:
LaserGrbl v3.8.0.0
Grbl v1.1f
Wrapper: UsbSerial
Win 10, Professional, 10.0.19041.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v2.0.50727\mscorlib.dll 2.0.50727.9151 (WinRelRS6.050727-9100)

TypeOf exception  [System.ObjectDisposedException]
Exception message [SafeHandle이 닫혀 있습니다.]
Exception source  [mscorlib], thread []
Exception method  [IntPtr SafeHandleC2NHelper(System.Object, IntPtr)]

   ----------- stack trace -----------
   위치: System.StubHelpers.StubHelpers.SafeHandleC2NHelper(Object pThis, IntPtr pCleanupWorkList)
   위치: Microsoft.Win32.UnsafeNativeMethods.WaitCommEvent(SafeFileHandle hFile, Int32* lpEvtMask, NativeOverlapped* lpOverlapped)
   위치: System.IO.Ports.SerialStream.EventLoopRunner.WaitForCommEvent()
   위치: System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   위치: System.Threading.ThreadHelper.ThreadStart()

BeatSlayer, 2020-11-20:
May be some bad usb cable or some other hardware problem.

Last time I had that error the power supply was "dying".

Cropwell-Rob, 2020-11-28:
Same problem for me, dropout after 5/6 seconds, UGS connects solidly so no hardware/driver/connection issue

arkypita, 2020-11-28:
> Same problem for me

Please try to be clearer. same problem for me doesn't mean much.
Add screenshot and describe what happens exactly can help.

> dropout after 5/6 seconds, UGS connects solidly so no hardware/driver/connection issue

LaserGRBL is optimized to reach the maximum speed of communication, this can put the most delicate/cheap hardware in trouble. I think UGS uses a much simple communication mechanism than LaserGRBL, so you cannot compare them.

In LaserGRBL faq http://lasergrbl.com/faq/ you can read all about how to solve connection issue, even setting lasergrbl to use a slower communication mechanism as workaround.

Cropwell-Rob, 2020-11-29:
OK, I was getting a very flaky connection to my win10 machine. The problem was that most of the time the USB serial port seemed to connect to the UART on my Cronosmaker only to drop out after a few seconds. I then tried UGS and it seemed to work OK, but subsequently has failed with the same problem. I tried all combinations of speeds and parity/stop bits etc and nothing worked.
Information and help from Cronosmaker seemed to be limited.
I then encountered an article on serial drivers and the CH340G chip and it suggested using the CH341SER driver. I deleted old drivers and installed this driver and the board now seems to connect reliably. I have yet to check that this is a consistent solution.
If it is then it could help a lot of people trying to connect cheap arduino clones using the CH340G chip.
LASERGRBL is my preferred software and I am happy to say, it looks as though I shall be able to use it.

I am sorry if I am not 'au fait' with modern terminology of windows error analysis, but my professional computing career started in 1966 and ended in 1999. Thirty years ago I could have bored you senseless with the internals of IBM360 series microcode, but today I am looking at a relaxing hobby and a means to an end by using Arduinos and C++.

Cropwell-Rob, 2020-11-29:
Here's the file I used for the drivers
[CH341SER.zip](https://github.com/arkypita/LaserGRBL/files/5612777/CH341SER.zip)
and here is the installed result
[image]

Cropwell-Rob, 2020-11-29:
After a reboot of my win10 machine, I am getting the dropouts (or failed connection?) again.  I am Lost !!

arkypita, 2020-11-29:
Hi @Cropwell-Rob 
Your computer knowledge is higher then 95% of LaserGRBL user eheheh

LaserGRBL is used by more than 50K users, for a total usage time of 15697 day 8 hour.
Only few user experience connection issue, otherwise it would be unusable.

However there are cases like yours, it is not the only one. It is really very difficult to diagnose these problems, what I can tell you is that 99% are external to LaserGRBL. Some people have solved it only by having a new controller board returned under warranty (defective controller).

Before thinking about a controller problem, check a few things: try another computer, another USB cable, another 12V power supply, move to another side of your house (fluorescent light, fridge, relais... they can disturb). Test your controller by disconnecting the motor cables, and disconnecting the laser (do it with the machine off).

Sending gcode with the stepper motor and laser disconnected is a good way to exclusively test the communication.

Remember to turn off the machine before reconnecting lasers and motors (motor drivers burn out if you unplug them while powered).

Cropwell-Rob, 2020-11-30:
Hi, My mainframe computer knowledge is archaic and irrelevant to today. All that matters to me now is the Arduino IDE gives me a great platform for hobby in my retirement.

I have spent too much time on this problem, but I can define that the problem is NOT within LaserGRBL. 
I have determined that the Cronosmaker board connects solidly and reliably when the off-line controller is not plugged in.
With the off-line controller connected then LaserGRBL appears to drop the serial link and will not jog the machine, which is OK by me. What is not good however, is that the facility to write GRBL parameters to the board is disabled. This may be by design, I cannot comment. I have found that PC files will still upload to LaserGRBL, so the connection to the PC has not actually been dropped.
So now I am looking for suitable workarounds. My intention is to mirror the method I use on my 3D printer, which has a Toshiba Flashair card with modifications to allow me to upload via Wifi and print offline.

I am bothered that the Cronosmaker hardware has no real accessible documentation as exemplified by the fact that it took me a long time to find silly info, like the voltage of the cooling fan connector. I fear that the pinout of the offline connector may remain a mystery.

Your advice is good, but not new to me.

Thanks,  Rob

closed 2020-12-16 by arkypita