arkypita/LaserGRBL #1175: Problem conecting to engraver

Union-man, 2020-11-21:
Trying to configure LaserGRBL but won't connect, the drop down only shows one choice, COM3 and my laptop doesn't have a com3.

StuartB4, 2020-11-21:
Have you installed a USB driver. I don't know what you machine is but there is a driver on the Tools menu in LaserGRBL. Your computer should automatically detect the engraver if it's switched on and you connect the USB lead.

Union-man, 2020-11-21:
Thanks, yes I did, I've tried reinstalling software, I've tried a different computer and managed to get the drop-down to show 2 COM ports and was able to match the COM to the COM in the device manager and still won't connect.
I've tried everything I know how to do.

I guess it's an Eleks, there's no brand name anywhere, the owners manual is a joke, I received a box with a thousand parts, no instructions and the little bit that is in the "manual" is  in Chinese, it was kind enough to have a picture of the wiring.
I did get it to do a couple things with Benbox but was hoping to find something better.
Thanks for trying.

arkypita, 2020-11-21:
If it work with benbox, then it won't work with LaserGRBL .
LaserGRBL works with machines that have grbl as controller firmware. Benbox use a different one.

http://lasergrbl.com/faq/
http://lasergrbl.com/usage/flash/

You can convert your controller from benbox to grbl, via the so called "firmware flash" procedure.
https://www.youtube.com/watch?v=9AecpJBzJDM

Union-man, 2020-11-22:
Hey thanks, I saw that flash thing but it gave some pretty dire warnings so I didn't try it.
Thanks a lot for your time.

closed 2020-12-16 by arkypita