arkypita/LaserGRBL #1176: LaserGRBL wont open

xtrmmatt, 2020-11-21:
I have used LaserGRBL for a while and now the program won't open.

When I open it it shows in the processes but doesnt open itself. I have unistalled and re-installed but no good.

I have windows 10 64 bit
 16GB RAM
i5-7200U

Framework is uptodate I have even installed .net 3.5 and no difference. I can see it in the resource monitor as suspended but I cannot end the process there as it comes with an error of access denied.

arkypita, 2020-11-21:
LaserGRBL settings folder is located here; C:\Users\<user name>\AppData\Roaming\LaserGRBL
Try delete setting folder. Let me know.

xtrmmatt, 2020-11-21:
[image]
I don't have a settings folder

arkypita, 2020-11-21:
LaserGRBL folder IS the setting folder.

xtrmmatt, 2020-11-21:
I deleted the lasergrbl folder, uninstalled the program, restarted my computer, reinstalled lasergrbl and the same thing. It shows in my processes but doesn't open.

xtrmmatt, 2020-11-21:
Thank you for your quick responses, it's a massive help

arkypita, 2020-11-21:
Any error recorded in sessionlog.txt (in setting folder)? Any error in windows event log?
Did you try run it ad administrator? Any antivirus that can disturb?

LaserGRBL only have it's install folder (C:\Program Files (x86)\LaserGRBL) and its setting folder. No other hidden files or registry entry.

arkypita, 2020-11-23:
News?