arkypita/LaserGRBL #1178: See Laser settings on UI

lottacustoms, 2020-11-23:
Hi,

it would be wonderfull seeing the actual setted speed and power of the laser at the UI.

Also it would be great to save and open  "project", that means the nc. code + the original setted speed and laser power.
So if you open again you don't have to remember the settings you did successfully last time.

Best regards
Marco

StuartB4, 2020-11-23:
`Also it would be great to save and open "project", that means the nc. code + the original setted speed and laser power.
So if you open again you don't have to remember the settings you did successfully last time.`

The speed and power settings are in the gcode. Just open the saved file and make a note of them.

lottacustoms, 2020-11-23:
Hi,

but anyware one of my chinese employees had already contact with them due they are on a chinese platform.
So if you need any information perhaps he can ask them directly.

Bit as already mentioned, it works fine with Laser GRBL for me.

And it works 2 cycles until it abords, but then simply start the job again and so no Problem.

Best regards
Marco

StuartB4, 2020-11-24:
I think you might have posted your reply in the wrong place, it has nothing to do with with your original request.

lottacustoms, 2020-11-24:
Oh thank you for the hint sorry :)

closed 2020-12-16 by arkypita