arkypita/LaserGRBL #1180: message when opening the lasergrbl program

espinosa777, 2020-11-27:
hello, when I open the lasergrbl program I get this error Does anyone know what the problem is?
thanks :

LaserGrbl v3.8.0.0
 v
Wrapper: UsbSerial
Win 10, Home, 10.0.19041.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v2.0.50727\mscorlib.dll 2.0.50727.9151 (WinRelRS6.050727-9100)

TypeOf exception  [System.Reflection.TargetInvocationException]
Exception message [Se produjo una excepción en el destino de la invocación.]
Exception source  [mscorlib], thread []
Exception method  [Void _SerializationInvoke(System.Object, System.SignatureStruct ByRef, System.Runtime.Serialization.SerializationInfo, System.Runtime.Serialization.StreamingContext)]

   ----------- stack trace -----------
   en System.RuntimeMethodHandle._SerializationInvoke(Object target, SignatureStruct& declaringTypeSig, SerializationInfo info, StreamingContext context)
   en System.Runtime.Serialization.ObjectManager.CompleteISerializableObject(Object obj, SerializationInfo info, StreamingContext context)
   en System.Runtime.Serialization.ObjectManager.FixupSpecialObject(ObjectHolder holder)
   en System.Runtime.Serialization.ObjectManager.DoFixups()
   en System.Runtime.Serialization.Formatters.Binary.ObjectReader.Deserialize(HeaderHandler handler, __BinaryParser serParser, Boolean fCheck, Boolean isCrossAppDomain, IMethodCallMessage methodCallMessage)
   en System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize(Stream serializationStream, HeaderHandler handler, Boolean fCheck, Boolean isCrossAppDomain, IMethodCallMessage methodCallMessage)
   en System.Resources.ResourceReader.DeserializeObject(Int32 typeIndex)
   en System.Resources.ResourceReader.LoadObjectV2(Int32 pos, ResourceTypeCode& typeCode)
   en System.Resources.ResourceReader.ResourceEnumerator.get_Entry()
   en System.Resources.ResourceReader.ResourceEnumerator.get_Current()
   en System.ComponentModel.ComponentResourceManager.FillResources(CultureInfo culture, ResourceSet& resourceSet)
   en System.ComponentModel.ComponentResourceManager.FillResources(CultureInfo culture, ResourceSet& resourceSet)
   en System.ComponentModel.ComponentResourceManager.ApplyResources(Object value, String objectName, CultureInfo culture)
   en LaserGRBL.ConnectLogForm.InitializeComponent()
   en LaserGRBL.ConnectLogForm..ctor()
   en LaserGRBL.MainForm.InitializeComponent()
   en LaserGRBL.MainForm..ctor()
   en LaserGRBL.Program.Main(String[] args)

Inner exception data
TypeOf exception  [System.ArgumentException]
Exception message [La fuente 'Consolas' no es compatible con el estilo 'Regular'.]
Exception source  [System.Drawing], thread []
Exception method  [Void CreateNativeFont()]

   ----------- stack trace -----------
   en System.Drawing.Font.CreateNativeFont()
   en System.Drawing.Font.Initialize(FontFamily family, Single emSize, FontStyle style, GraphicsUnit unit, Byte gdiCharSet, Boolean gdiVerticalFont)
   en System.Drawing.Font..ctor(SerializationInfo info, StreamingContext context)

LaserGrbl v3.8.0.0
 v
Wrapper: UsbSerial
Win 10, Home, 10.0.19041.0 (7)
CLR: C:\Windows\Microsoft.NET\Framework64\v2.0.50727\mscorlib.dll 2.0.50727.9151 (WinRelRS6.050727-9100)

TypeOf exception  [System.Reflection.TargetInvocationException]
Exception message [Se produjo una excepción en el destino de la invocación.]
Exception source  [mscorlib], thread []
Exception method  [Void _SerializationInvoke(System.Object, System.SignatureStruct ByRef, System.Runtime.Serialization.SerializationInfo, System.Runtime.Serialization.StreamingContext)]

   ----------- stack trace -----------
   en System.RuntimeMethodHandle._SerializationInvoke(Object target, SignatureStruct& declaringTypeSig, SerializationInfo info, StreamingContext context)
   en System.Runtime.Serialization.ObjectManager.CompleteISerializableObject(Object obj, SerializationInfo info, StreamingContext context)
   en System.Runtime.Serialization.ObjectManager.FixupSpecialObject(ObjectHolder holder)
   en System.Runtime.Serialization.ObjectManager.DoFixups()
   en System.Runtime.Serialization.Formatters.Binary.ObjectReader.Deserialize(HeaderHandler handler, __BinaryParser serParser, Boolean fCheck, Boolean isCrossAppDomain, IMethodCallMessage methodCallMessage)
   en System.Runtime.Serialization.Formatters.Binary.BinaryFormatter.Deserialize(Stream serializationStream, HeaderHandler handler, Boolean fCheck, Boolean isCrossAppDomain, IMethodCallMessage methodCallMessage)
   en System.Resources.ResourceReader.DeserializeObject(Int32 typeIndex)
   en System.Resources.ResourceReader.LoadObjectV2(Int32 pos, ResourceTypeCode& typeCode)
   en System.Resources.ResourceReader.ResourceEnumerator.get_Entry()
   en System.Resources.ResourceReader.ResourceEnumerator.get_Current()
   en System.ComponentModel.ComponentResourceManager.FillResources(CultureInfo culture, ResourceSet& resourceSet)
   en System.ComponentModel.ComponentResourceManager.FillResources(CultureInfo culture, ResourceSet& resourceSet)
   en System.ComponentModel.ComponentResourceManager.ApplyResources(Object value, String objectName, CultureInfo culture)
   en LaserGRBL.ConnectLogForm.InitializeComponent()
   en LaserGRBL.ConnectLogForm..ctor()
   en LaserGRBL.MainForm.InitializeComponent()
   en LaserGRBL.MainForm..ctor()
   en LaserGRBL.Program.Main(String[] args)

Inner exception data
TypeOf exception  [System.ArgumentException]
Exception message [La fuente 'Consolas' no es compatible con el estilo 'Regular'.]
Exception source  [System.Drawing], thread []
Exception method  [Void CreateNativeFont()]

   ----------- stack trace -----------
   en System.Drawing.Font.CreateNativeFont()
   en System.Drawing.Font.Initialize(FontFamily family, Single emSize, FontStyle style, GraphicsUnit unit, Byte gdiCharSet, Boolean gdiVerticalFont)
   en System.Drawing.Font..ctor(SerializationInfo info, StreamingContext context)

arkypita, 2020-11-27:
> Exception message [La fuente 'Consolas' no es compatible con el estilo 'Regular'.]

Look like you have issue with Font on your computer.
Follow this link:

https://answers.microsoft.com/en-us/windows/forum/windows_10-start/some-fonts-are-missing-after-upgrade/95839dfa-0df2-4bc0-875a-fd6b57e61fe4

https://social.technet.microsoft.com/Forums/windows/en-US/4a1b0d78-bed3-465b-be38-859b4180a01b/missing-fonts-after-creators-update-installed?forum=win10itprogeneral

www.google.com/search?q=createnativefont+%5BFont+%27Consolas%27+does+not+support+style+%27Regular%27.%5D&oq=createnativefont+%5BFont+%27Consolas%27+does+not+support+style+%27Regular%27.%5D

arkypita, 2020-11-27:
News?

espinosa777, 2020-11-27:
No se ha resuelto el problema 😅